realsenseai/librealsense #2472: Application hangs when trying to close file replay pipeline

blackccpie, 2018-10-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.10.3 | 
| Operating System & Version |   Win 10  |                                      | 
| Platform                                 | PC  |
| SDK Version                            |  2.16.1                          | 
| Language                            |  C++                          | 

### Issue Description
It seems like since I upgraded to sdk version 2.16.1, my application sometimes hang when trying to close a replay pipeline (I'm playing bag files that have been recorded with a former sdk version). I'm starting to investigate, and it appears I get stuck in the `pipeline::unsafe_stop()` method, and more precisely when calling `_active_profile->_multistream.stop()`. This looks like a deadlock case, but I'm not really sure.

I'm going further in my investigations, but if someone has encountered the same problem and has a solution I'll be glad to hear.

Regards,

Albert

blackccpie, 2018-10-03:
EDIT : seems like I'm permanently stuck in the while loop at line 224 of `concurrency.h`, in the dispatcher thread.

blackccpie, 2018-10-04:
Hi @RealSense-Customer-Engineering, my previous working version of the sdk was **2.15.0.**
I also forgot to mention I was using the newly functional realtime mode during my replay.
I will collect the logs asap.

blackccpie, 2018-10-04:
When the app exits fine get:

```text
04/10 11:23:53,442 DEBUG [10696] (ros_reader.h:60) Next message is a frame
 04/10 11:23:53,442 DEBUG [10696] (ros_reader.h:435) Trying to create an image frame from message
 04/10 11:23:53,442 INFO [2896] (archive.cpp:316) The user was holding on to 1 frames after stream 0x0000020197832500 stopped
 04/10 11:23:53,442 INFO [2896] (archive.cpp:316) The user was holding on to 1 frames after stream 0x0000020197832500 stopped
 04/10 11:23:53,442 INFO [2896] (archive.cpp:316) The user was holding on to 1 frames after stream 0x0000020196C603C0 stopped
 04/10 11:23:53,442 INFO [2896] (archive.cpp:316) The user was holding on to 1 frames after stream 0x0000020196C603C0 stopped
 04/10 11:23:53,442 INFO [2896] (archive.cpp:316) The user was holding on to 1 frames after stream 0x0000020196C603C0 stopped
 04/10 11:23:53,442 INFO [2896] (archive.cpp:316) The user was holding on to 1 frames after stream 0x0000020196C603C0 stopped
 04/10 11:23:53,442 DEBUG [2896] (archive.cpp:249) CallbackFinished,Depth,316,DispatchedAt,1.53865e+12
 04/10 11:23:53,442 DEBUG [2896] (archive.cpp:256) Frame Callback [Depth#316] overdue. (Duration: 35261.6ms, FPS: 30, Max Duration: 32ms)
 04/10 11:23:53,442 INFO [2896] (archive.cpp:326) All frames from stream 0x0000020196C603C0 are now released by the user
 04/10 11:23:53,442 DEBUG [2896] (archive.cpp:249) CallbackFinished,Depth,5af,DispatchedAt,1.53865e+12
 04/10 11:23:53,442 DEBUG [2896] (archive.cpp:256) Frame Callback [Depth#1455] overdue. (Duration: 1.53865e+12ms, FPS: 30, Max Duration: 32ms)
 04/10 11:23:53,442 INFO [2896] (archive.cpp:326) All frames from stream 0x000002019FC77DC0 are now released by the user
 04/10 11:23:53,442 DEBUG [2896] (archive.cpp:249) CallbackFinished,Color,0,DispatchedAt,1.53865e+12
 04/10 11:23:53,442 INFO [2896] (archive.cpp:326) All frames from stream 0x0000020197533020 are now released by the user
 04/10 11:23:53,442 DEBUG [2896] (archive.cpp:249) CallbackFinished,Depth,5af,DispatchedAt,1.53865e+12
 04/10 11:23:53,442 DEBUG [2896] (archive.cpp:256) Frame Callback [Depth#1455] overdue. (Duration: 1.53865e+12ms, FPS: 30, Max Duration: 32ms)
 04/10 11:23:53,442 INFO [2896] (archive.cpp:326) All frames from stream 0x0000020197762DC0 are now released by the user
 04/10 11:23:53,442 DEBUG [2896] (archive.cpp:249) CallbackFinished,Color,5a3,DispatchedAt,1.53865e+12
 04/10 11:23:53,442 DEBUG [2896] (archive.cpp:256) Frame Callback [Color#1443] overdue. (Duration: 1.53865e+12ms, FPS: 30, Max Duration: 32ms)
 04/10 11:23:53,442 INFO [2896] (archive.cpp:326) All frames from stream 0x00000201975F3030 are now released by the user
 04/10 11:23:53,442 DEBUG [2896] (archive.cpp:249) CallbackFinished,Depth,0,DispatchedAt,1.53865e+12
 04/10 11:23:53,442 DEBUG [2896] (archive.cpp:256) Frame Callback [Depth#0] overdue. (Duration: 1.53865e+12ms, FPS: 30, Max Duration: 32ms)
 04/10 11:23:53,442 INFO [2896] (archive.cpp:326) All frames from stream 0x0000020197832500 are now released by the user
 04/10 11:23:53,442 DEBUG [10696] (ros_reader.h:476) Created image frame: 0/1/2/0 500x2d0 5
 04/10 11:23:53,442 DEBUG [10696] (playback_device.cpp:370) Updating Time Base... m_base_sys_time 44f9cbf5c3b3 m_base_timestamp 16f
 04/10 11:23:53,442 DEBUG [10696] (playback_device.cpp:578) Dispatching frame 0/1/2/0
 04/10 11:23:53,457 DEBUG [10696] (archive.cpp:249) CallbackFinished,Color,77,DispatchedAt,1.53865e+12
 04/10 11:23:53,457 DEBUG [10696] (archive.cpp:256) Frame Callback [Color#119] overdue. (Duration: 1.53865e+12ms, FPS: 30, Max Duration: 32ms)
 04/10 11:23:53,457 ERROR [10696] (playback_device.cpp:470) Failed to read next frame from file: invalid map<K, T> key
 04/10 11:23:53,457 DEBUG [10696] (playback_sensor.cpp:141) Stop sensor 0
 04/10 11:23:53,457 DEBUG [10696] (playback_sensor.cpp:141) Stop sensor 1
 04/10 11:23:53,504 DEBUG [10696] (ros_file_format.h:430) RegexTopicQuery with expression: /device_0/sensor_(\d)+/info
 04/10 11:23:53,504 DEBUG [10696] (ros_file_format.h:430) RegexTopicQuery with expression: /device_0/sensor_0/(\w)+_(\d)+/info
 04/10 11:23:53,504 DEBUG [10696] (ros_file_format.h:430) RegexTopicQuery with expression: /device_0/sensor_0/Depth_0/tf
 04/10 11:23:53,504 DEBUG [10696] (ros_file_format.h:430) RegexTopicQuery with expression: /device_0/sensor_1/(\w)+_(\d)+/info
 04/10 11:23:53,520 DEBUG [10696] (ros_file_format.h:430) RegexTopicQuery with expression: /device_0/sensor_1/Color_0/tf
 04/10 11:23:53,520 DEBUG [10696] (playback_device.cpp:625) Catching up
```

When the app hangs I get:

```text
04/10 11:19:21,747 DEBUG [4072] (ros_reader.h:60) Next message is a frame
 04/10 11:19:21,747 DEBUG [12792] (sync.cpp:99) I Depth--> Depth 122, 1536249932095.207520

 04/10 11:19:21,747 DEBUG [12792] (sync.cpp:258) SYNC (TS: )--> Depth 122 1536249932095.207520 

 04/10 11:19:21,747 DEBUG [4072] (ros_reader.h:435) Trying to create an image frame from message
 04/10 11:19:21,747 DEBUG [12792] (sync.cpp:516) fps 0 Depth 122 1536249932095.207520 
 04/10 11:19:21,747 DEBUG [12792] (sync.cpp:529) (TS: )Depth 122 1536249932095.207520 fps 30 gap 33.3333 next_expected: 1.53625e+12
 04/10 11:19:21,747 DEBUG [12792] (sync.cpp:516) fps 0 Depth 122 1536249932095.207520 
 04/10 11:19:21,747 DEBUG [12792] (sync.cpp:516) fps 0 Color 121 1536249932093.854492 
 04/10 11:19:21,747 DEBUG [12792] (syncer-processing-block.cpp:28) SYNCED: Color 121, 1536249932093.854492 Depth 122, 1536249932095.207520 
[... cut ...]
 04/10 11:19:21,924 DEBUG [3012] (archive.cpp:249) CallbackFinished,Depth,123,DispatchedAt,1.53864e+12
 04/10 11:19:21,924 DEBUG [3012] (archive.cpp:256) Frame Callback [Depth#123] overdue. (Duration: 1.53864e+12ms, FPS: 30, Max Duration: 32ms)
 04/10 11:19:21,924 DEBUG [3012] (archive.cpp:249) CallbackFinished,Depth,124,DispatchedAt,1.53864e+12
 04/10 11:19:21,924 DEBUG [3012] (archive.cpp:256) Frame Callback [Depth#124] overdue. (Duration: 1.53864e+12ms, FPS: 30, Max Duration: 32ms)
```

In the latter case I don't get the `Stop sensor 1` log.

dorodnic, 2018-10-04:
@matkatz please take a look

matkatz, 2018-12-11:
Hi @blackccpie 
Regarding:
> I'm using the pipeline::poll_for_frames method to query my frames, but it sometimes returns false even when the stream has not ended. In non-realtime mode, I would have expected to get all the frames of the file, without having poll_for_frames to return false, isn't that right?

When working in non-realtime mode, `poll_for_frames` is still limited by the file reading speed from the HDD, so it is possible for `poll_for_frames` to return false.

Regarding the hang issue:
#2849 fix an issue that may be related to the hang that you saw, it is currently merged to the development branch and will be part of the next release.

blackccpie, 2018-12-19:
Hi @matkatz , thanks for the explanation, it's clearer for me now. 
I'll have a give a try with the next release to see if my problem is solved.
By the way do you have an opinion concerning my question about the playback stop condition in [this comment](https://github.com/IntelRealSense/librealsense/issues/1579#issuecomment-430163908)?

Regards

RealSenseCustomerSupport, 2019-02-15:
----------------------------------------------

@blackccpie Could you please have a try with latest release to see if still have issue? Thanks!

blackccpie, 2019-02-18:
Hi @RealSenseCustomerSupport, in fact since _matkatz_ answer I was waiting for a **release** version to spent some time on upgrading/testing, but since then there only have been **pre-releases.** Do you know when the next major release version will be out? thx

blackccpie, 2019-02-21:
@RealSenseCustomerSupport : I took some time to test the pre-release v2.18.1, and the issue seems fixed. Anyway I'm still interested to know when the next official release will be available.

Regards

closed 2019-03-01 by blackccpie