realsenseai/librealsense #2473: rs2_pipeline_start_with_config can't resolve requests in C when trying to stream color and depth

matlabninja, 2018-10-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.10.3.0 | 
| Operating System & Version | Linux (Ubuntu 16.04) | 
| Kernel Version (Linux Only)    |  4.4.134 | 
| Platform                                 | HP Z Book PC |
| SDK Version                            | 2.16.0 | 
| Language                            | C | 
| Segment			| Robot | 

### Issue Description
I am attempting to stream both color and depth information from the device in a C program that I have modeled after the rs-depth example.  The code is inside a static bool function that is called from another function.  When executing rs2_start_pipeline with config, I receive the following error: 
rs_error was raised when calling rs2_pipeline_start_with_config(pipe:0x7f9a9800aea0, config:0x7f9a9800a140):
    Couldn't resolve requests

Minimal code to reproduce (removed error checking and printing from the example, which is what printed the error):
#include <librealsense2/rs.h>
#include <librealsense2/h/rs_pipeline.h>
#include <librealsense2/h/rs_option.h>
#include <librealsense2/h/rs_frame.h>
#include "example.h" //The example.h copied from the C examples in this repo

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
//                                     These parameters are reconfigurable                                        //
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
#define STREAM_DEPTH    RS2_STREAM_DEPTH  // rs2_stream is a types of data provided by RealSense device           //
#define FORMAT_DEPTH    RS2_FORMAT_Z16    // rs2_format is identifies how binary data is encoded within a frame   //
#define STREAM_COLOR    RS2_STREAM_COLOR  // rs2_stream is a types of data provided by RealSense device           //
#define FORMAT_COLOR    RS2_FORMAT_RGB8   // rs2_format is identifies how binary data is encoded within a frame   //
#define WIDTH           1280              // Defines the number of columns for each frame                         //
#define HEIGHT          720               // Defines the number of lines for each frame                           //
#define FPS             20                // Defines the rate of frames per second                                //
#define STREAM_INDEX    -1                // Defines the stream index, used for multiple streams of the same type //
//////////////////////////////////////////////////////////////////////////////////////////////////////////////////// 

static bool realsense_handler(void)
{
	rs2_frame* frames;
	rs2_context* ctx;
	rs2_device_list* device_list;
	rs2_device* dev;
	rs2_pipeline* pipeline;
	rs2_config* config;
	rs2_pipeline_profile* pipeline_profile;
	
	// Open objects for talking to the realsense camera
	// Realsense error object
	rs2_error* e = 0;
	// Contect object that will own the realsense h/w stuff
	ctx = rs2_create_context(RS2_API_VERSION, &e);

	/* Get a list of all the connected devices. */
	// The returned object should be released with rs2_delete_device_list(...)
	device_list = rs2_query_devices(ctx, &e);

	int dev_count = rs2_get_device_count(device_list, &e);

	// Get the first connected device
	// The returned object should be released with rs2_delete_device(...)
	dev = rs2_create_device(device_list, 0, &e);
	print_device_info(dev);

	// Get quantization value of depth image
	float depth_scale = get_depth_unit_value(dev);
	
	// Open pipeline to realsense device
	pipeline = rs2_create_pipeline(ctx,&e);

	// Get the device config, necessary for starting a stream
	config = rs2_create_config(&e);

	// Set streaming from device
	rs2_config_enable_stream(config,STREAM_DEPTH,STREAM_INDEX,WIDTH,HEIGHT,FORMAT_DEPTH,FPS,&e);
	rs2_config_enable_stream(config,STREAM_COLOR,STREAM_INDEX,WIDTH,HEIGHT,FORMAT_COLOR,FPS,&e);

	// Start streaming pipeline
	pipeline_profile = rs2_pipeline_start_with_config(pipeline, config, &e);

	// Do other stuff

Output:
Using device 0, an Intel RealSense D435
    Serial number: 745412070436
    Firmware version: 05.10.03.00

rs_error was raised when calling rs2_pipeline_start_with_config(pipe:0x7f9a9800aea0, config:0x7f9a9800a140):
    Couldn't resolve requests

ev-mp, 2018-10-03:
@matlabninja ,
There is a typo in the line
>#define FPS 20 // Defines the rate of frames per second //

which is not a valid FPS for D435. Switch to 30 or 15, if you're using USB2, and retry.

To obtain the full range of supported fps modes and resolutions you can run  `rs-enumerate-devices`

matlabninja, 2018-10-03:
@ev-mp That'll do it, thanks!

closed 2018-10-03 by matlabninja

mmubeen-6, 2019-03-07:
Hey @matlabninja and @ev-mp.
I have extended the code that you provided in this thread, the new code is given below: -

```
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2017 Intel Corporation. All Rights Reserved.

/* Include the librealsense C header files */
#include <librealsense2/rs.h>
#include <librealsense2/h/rs_pipeline.h>
#include <librealsense2/h/rs_frame.h>

#include <stdlib.h>
#include <stdint.h>
#include <stdio.h>
#include "../example.h"

#include <opencv2/opencv.hpp>            // C++
#include <opencv2/highgui/highgui_c.h>
#include <opencv2/core/core_c.h>
#include <opencv2/core/version.hpp>
#include <opencv2/imgproc/imgproc_c.h>
#include <opencv2/videoio/videoio_c.h>

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
//                                     These parameters are reconfigurable                                        //
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
#define STREAM_COLOR    RS2_STREAM_COLOR  // rs2_stream is a types of data provided by RealSense device           //
#define FORMAT_COLOR    RS2_FORMAT_RGB8   // rs2_format is identifies how binary data is encoded within a frame   //
#define STREAM_DEPTH    RS2_STREAM_DEPTH  // rs2_stream is a types of data provided by RealSense device //
#define FORMAT_DEPTH    RS2_FORMAT_Z16    // rs2_format is identifies how binary data is encoded within a frame //
#define WIDTH           640               // Defines the number of columns for each frame                         //
#define HEIGHT          480               // Defines the number of lines for each frame                           //
#define FPS             30                // Defines the rate of frames per second                                //
#define STREAM_INDEX    -1                // Defines the stream index, used for multiple streams of the same type //
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

int main()
{
    rs2_error* e = 0;

    // Create a context object. This object owns the handles to all connected realsense devices.
    // The returned object should be released with rs2_delete_context(...)
    rs2_context* ctx = rs2_create_context(RS2_API_VERSION, &e);
    check_error(e);

    /* Get a list of all the connected devices. */
    // The returned object should be released with rs2_delete_device_list(...)
    rs2_device_list* device_list = rs2_query_devices(ctx, &e);
    check_error(e);

    int dev_count = rs2_get_device_count(device_list, &e);
    check_error(e);
    printf("There are %d connected RealSense devices.\n", dev_count);
    if (0 == dev_count)
        return EXIT_FAILURE;

    // Get the first connected device
    // The returned object should be released with rs2_delete_device(...)
    rs2_device* dev = rs2_create_device(device_list, 0, &e);
    check_error(e);

    print_device_info(dev);

    // Create a pipeline to configure, start and stop camera streaming
    // The returned object should be released with rs2_delete_pipeline(...)
    rs2_pipeline* pipeline =  rs2_create_pipeline(ctx, &e);
    check_error(e);

    // Create a config instance, used to specify hardware configuration
[... 67 lines cut ...]
        check_error(e);

        rs2_release_frame(frame_color);
        rs2_release_frame(frame_depth);
        rs2_release_frame(frames);
    }

    // Stop the pipeline streaming
    rs2_pipeline_stop(pipeline, &e);
    check_error(e);

    // Release resources
    rs2_delete_pipeline_profile(pipeline_profile);
    rs2_delete_config(config);
    rs2_delete_pipeline(pipeline);
    rs2_delete_device(dev);
    rs2_delete_device_list(device_list);
    rs2_delete_context(ctx);

    return EXIT_SUCCESS;
}
```

But right now, I am unable to run this code, only the RGB frame is received, and even that is not working properly.
Any help would be appreciated.