realsenseai/librealsense #2475: Framerate problems in RealSense Viewer after version 2.13.0

xtrawurst, 2018-10-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 series | 
| Firmware Version                   | any | 
| Operating System & Version |  Win10 | 

### Issue Description
To reproduce:
1. Plug in a RS D415 or D435
2. Start RS Viewer 2.13.0 and turn on color streaming. Should be 30 fps.
3. Close RS Viewer 2.13.0
4. Open RS Viewer 2.14.0 or later and turn on color streaming.
5. Framerate is not 30 fps.

Similar issues in the Viewer with depth or combined depth + color.

Fortunately when we use libRS in our own app we don't see these issues and get full frame rate.

RealSense-Customer-Engineering, 2018-10-09:
[Realsense Customer Engineering Team Comment]
Hi @xtrawurst,

Did you build the realsense-viewer from the source? I didn't observe this issue on my configuration.
You can also try the pre-built realsense-viewer to see how it works.

xtrawurst, 2018-10-09:
@RealSense-Customer-Engineering We used the compiled builds that can be found [here](https://github.com/IntelRealSense/librealsense/releases).

RealSense-Customer-Engineering, 2018-10-12:
[Realsense Customer Engineering Team Comment]
Hi @xtrawurst,

Please help enable the log "set LRS_LOG_LEVEL=DEBUG" first in the command line.
Once the issue reproduced, please send us the log file which should be the same location with your realsense-viewer.

RealSense-Customer-Engineering, 2018-11-06:
[Realsense Customer Engineering Team Comment]
Hi @xtrawurst,

Is the issue still there for further debug?

xtrawurst, 2018-11-12:
I can confirm that this is no longer an issue in 2.16.3.

closed 2018-11-12 by xtrawurst