realsenseai/librealsense #2478: Rasbain os support (OS for Rasberry PI)

ShettyHarapanahalli, 2018-10-04:
Does real sense support the Rasbain OS.

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux
| Kernel Version (Linux Only)    |   
| Platform                                 | Raspberry Pi|
| SDK Version                            |  {  2.x }                          | 
| Language                            |  {C/python}                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Does real sense support the Rasbain OS.
I have to use ROS/ tensorflow and real sense for my project. Do we have support for realsense in rasbain.

MartyG-RealSense, 2018-10-04:
I do not know of support for Raspbian specifically,  However, the link below has instructions for setting up the SDK on any version of Linux.

https://github.com/IntelRealSense/librealsense/issues/1132#issuecomment-363572666

Alternatively, Ubuntu MATE could be used with the Raspberry Pi, as described in the guide to setting up a 400 Series camera on the Pi.

https://github.com/IntelRealSense/librealsense/blob/v2.8.2/doc/RaspberryPi3.md

dorodnic, 2018-10-07:
Hi @ShettyHarapanahalli 
Specifically for Raspberry Pi integration we chose to go with Ubuntu Mate, because it is a bit closer to the "standard" Linux configuration which is for us Ubuntu 16.04 LTS. 
You can certainly build the SDK from source on Rasbain, but getting all the right tool-chain (GCC 5, GLFW, CMake, etc..) can take some time. 
At the hardware integration level, the key factor is the kernel version of your distribution. 
You should get streaming out of the box (with kernel > 4.4), and hardware timestamps if kernel version >= 4.16.

ShettyHarapanahalli, 2018-10-08:
Thanks for the response, I am able to use libsense in here :)

closed 2018-10-08 by ShettyHarapanahalli

ShettyHarapanahalli, 2018-10-16:
Just for reference, I see that the max [more than 905] CPU is taken by real sense simple application only. 
So we can not have any other processing included reliably onto PI [pi 3 is used] as its already overloaded.