realsenseai/librealsense #2479: D415 / D435 on Android: Color auto-exposure not working

xtrawurst, 2018-10-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 series | 
| Firmware Version                   | any | 
| Operating System & Version | Android 8.0  | 
| SDK Version                            |  2.16.0 | 
| Segment			|  Tablet | 

### Issue Description
I am able to use librealsense as a library on Android with our application. However it looks like the RGB sensor auto exposure is not working. The color image exposure remains static.

I can set and query the AE flags through `rs2::sensor::set/get_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE)` and the flags are correctly set/queried, as one would expect if working, however the sensor behaviour just isn't reflecting it.

As far as I can see the depth sensor AE seems to be working correctly.

Is this a known issue?

dorodnic, 2018-10-04:
Hi @xtrawurst 
This is not a known-issue, we need to look more closely into it.

xtrawurst, 2018-10-04:
I should add that we are using a USB-C port and request depth at 1280x720 and RGB at 640x360.

AnnaRomanov, 2018-10-09:
Hi @xtrawurst, we made some changes in the libuvc backend: #2490
The problem is not reproduced on my device after the changes, let us know if you still have this issue.

xtrawurst, 2018-10-09:
@AnnaRomanov Great, thanks! I will try this out.

xtrawurst, 2018-10-10:
@AnnaRomanov @dorodnic 
This looks good! We no longer see the issue.