realsenseai/librealsense #2480: Corrupted right infrared frame

WallyBender, 2018-10-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415| 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |  Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-34-generic | 
| Platform                                 | PC |
| SDK Version                            |  2.0  | 
| Language                            |  C++/Python                          | 
| Segment			| Robot | 

### Issue Description
We are using C++ SDK to write depth images and found that sometimes depth images are corrupted. With some debugging, we found out that one of the infrared frames is corrupted - it is very noisy and the left edge has corrupted data. 

When both the frames are not corrupted, the final depth frame is good when compared to the corrupted case.

[image]

[image]

dorodnic, 2018-10-04:
Hi @WallyBender 
First thing I'd recommend is updating to latest firmware.
Next, we'd love to understand how this behaviour can be reproduced. Can it be a single malfunctioning camera?

MartyG-RealSense, 2018-10-04:
Dorodnic beat me to it with the suggestion to update your firmware.  :)

The camera firmware that you have specified you are using (5.8.15) is a very early version of the firmware that dates back to December 2017.  There have been six versions of the firmware since then that have added extensive features and fixes.

Would it be possible to update your camera to the current firmware please and see if your problem persists?  

The link below gives details of how to update the firmware on Linux.

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Linux-RealSense-D400-DFU-Guide.pdf

RealSense-Customer-Engineering, 2018-10-16:
[Realsense Customer Engineering Team Comment]
1. As dorodnic suggested, please update to latest FW

2. Did this issue happen all the time or sometime?

3. Please capture the 3 images: depth, left IR, right IR

RealSense-Customer-Engineering, 2018-11-02:
[Realsense Customer Engineering Team Comment]
Hi WallyBender,
Do you still have this issue?

kerkildc, 2018-11-02:
Hi @RealSense-Customer-Engineering,

Yes, we do still have these issues (I am working with WallyBender in the same organization). To answer your questions:

1. We have upgraded both librealsense and the FW to the latest (Linux version stays the same). 
2. This appears to happen only sometimes (possibly with only some cameras) - we use multiple realsense cameras in the same hardware, they do not happen consistently across all the cameras.
3. We will grab some images when we have chances and post them here.

Thanks for the response.

waspinator, 2018-11-25:
I have a similar issue with the D435. I have two of them and both have the same issue.

Sometimes the left, and sometimes the right camera will become corrupted with static lines. They fix themselves after a while, and then break again. 

[image]

RealSense-Customer-Engineering, 2018-11-25:
[Realsense Customer Engineering Team Comment]
What does the depth image look like when this happen for left/right? Could you capture the depth image?

waspinator, 2018-11-25:
@RealSense-Customer-Engineering 

Yes, I can capture depth images. But they aren't any good.

[image]

RealSense-Customer-Engineering, 2018-11-28:
[Realsense Customer Engineering Team Comment]
Do you use RealSense Viewer? Or use your own viewer?

waspinator, 2018-11-28:
I used the realsense viewer to capture those examples.

RealSense-Customer-Engineering, 2018-11-29:
[Realsense Customer Engineering Team Comment]
This seems HW issue...
When did you purchase the camera? Could you please send us the serial numbers?

waspinator, 2018-12-04:
- 814412070156
- 814412072111

I ordered them early this year.

RealSense-Customer-Engineering, 2018-12-07:
[Realsense Customer Engineering Team Comment]
Could you please try these two cameras on a Windows 10 PC to check if they have the same issue?

waspinator, 2018-12-07:
it's hard to troubleshoot that way, because the data corruption isn't easily reproducible. I had the camera taking time-lapse images over the course of several months, and ~30% of the images were broken. Without doing anything the cameras would recover for a while and then fail again.

Just plugging the camera in works on both Windows 10 and Ubuntu 16.04.

RealSense-Customer-Engineering, 2018-12-07:
[Realsense Customer Engineering Team Comment]
Could you please  try the latest FW - 5.10.6?

closed 2019-01-02 by RealSense-Customer-Engineering

MartyG-RealSense, 2020-08-04:
Hi @waspinator  The information below is unrelated to the current case.  I saw your ROS case about getting information from a .raw file.  As I can't post on that case as I'm not assigned to it, I thought I would share my findings in a comment here instead as a one-off comment in order to share the info with you.

***********

Hi @waspinator   Apologies for the delay in responding.  A range of methods for utilising raw data without ROS are suggested below.

***********************

One RealSense user took the approach of creating a script to convert the .raw file into a .ply point cloud using cv::mat:

https://github.com/IntelRealSense/librealsense/issues/3286#issuecomment-465648074

Another RealSense user provided a Python script that saves color as a PNG and saves depth as an array of scaled matrices saved as an .npy file:

https://github.com/IntelRealSense/librealsense/issues/4934#issuecomment-537705225

Another approach for getting the depth data from a .raw file is suggested by a RealSense team member here:

https://github.com/IntelRealSense/librealsense/issues/2231#issuecomment-415451870

*************

You may also be interested in the Python / OpenCV measuring project in the link below.  It was originally a Python conversion of the SDK's **rs-measure** C++ example but was later expanded into a larger project.  The original measure code can still be accessed as a standalone script within the project's folders though.

**Full project**
https://github.com/soarwing52/RealsensePython

**Standalone measure script**
https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/measure_new.py