realsenseai/librealsense #2483: how do I replicate rs-pcl.cpp in python?

phojjat, 2018-10-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I am trying to replicate rs-pcl.cpp in py realsense. I have python-pcl installed and need to convert rs pointcloud to pcl point cloud. The function to do this in c is 

pcl_ptr points_to_pcl(const rs2::points& points)
{
    pcl_ptr cloud(new pcl::PointCloud<pcl::PointXYZ>);

    auto sp = points.get_profile().as<rs2::video_stream_profile>();
    cloud->width = sp.width();
    cloud->height = sp.height();
    cloud->is_dense = false;
    cloud->points.resize(points.size());
    auto ptr = points.get_vertices();
    for (auto& p : cloud->points)
    {
        p.x = ptr->x;
        p.y = ptr->y;
        p.z = ptr->z;
        ptr++;
    }

    return cloud;
}

So I try to replicate this in python as follows:

def point_to_pcl(pcl_point,point): 
    pcl_point=pcl.PointCloud()
    
    #sp=rs.stream_profile.as_video_stream_profile 
    sp=point.get_profile().as_video_stream_profile()
    pcl_point.width = sp.width()
    pcl_point.height = sp.height()
    pcl_point.is_dense = false
    pcl_point.resize(point.size())
    ptr = point.get_vertices()
    pcl_point.points.size()
    for p in range(0, pcl_point.points.size()):
        p.x = ptr.x
        p.y = ptr.y
        p.z = ptr.z
        ptr=ptr+1

However I keep getting: RuntimeError: null pointer passed for argument "frame_ref" at the line " sp=point.get_profile().as_video_stream_profile()". Any help to resolve this is much appreciated!

MartyG-RealSense, 2018-10-05:
The user @baptiste-mnh has previously recommended a python binding of the PCL library called python-pcl.

https://github.com/strawlab/python-pcl

phojjat, 2018-10-05:
I have python-pcl installed bu their point_grabber is not correctly compiled so I cannot use it readily. I need to convert the realsense pointcloud to pcl to be able to use the pcl filters. 

I actually managed to pass the null pointer error and now I am getting an attribute error at the line below saying "attribute 'width' of 'pcl._pcl.PointCloud' objects is not writable" 

so I wonder if this is even possible

MartyG-RealSense, 2018-10-05:
Is it absolutely necessary for you to use PCL to generate a point cloud please, or are you able to consider python-based alternatives?

phojjat, 2018-10-05:
In order to run python pcl filters, I think I need the point cloud to be a
PCL point cloud. I am not sure if there is a way to construct a point cloud
that is identical to a pcl point cloud. I wasn't able to run pcl functions
on just rs.pointcloud so just assumed I need a pcl point cloud

phojjat, 2018-10-05:
What I want to do is ground plane segmentation from a point cloud. I know python-pcl has a RANSAC-based algorithm and initially though that to be the right choice. If you know of other platforms out there that work better with realsense please let me know

MartyG-RealSense, 2018-10-05:
Dorodnic, RealSense's SDK Manager, is an advocate of a very versatile Python point cloud system called Pyntcloud.

https://github.com/daavoo/pyntcloud

phojjat, 2018-10-09:
> Hi @phojjat <https://github.com/phojjat>
> Could you please share more information about how point_to_pcl function
> is being invoked?

Hello,

pc=rs.pointcloud()
pc.map_to(aligned_depth_frame)
points = pc.calculate(aligned_depth_frame)
pcl_points=pcl.PointCloud()
point_to_pcl(pcl_points,points)

phojjat, 2018-10-10:
So I installed pyntcloud and ran the tests and everything seemed okay, but now the my pyrealsense2 seems to be broken:

 import pyrealsense2
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'pyrealsense2'

I tried recloning/reinstalling libreasense, but didn't fix it.

Is there a known problem with pyntcloud and librealsense2 that you can discuss? Otherwise do you know how I can fix this?

Thanks!

RealSense-Customer-Engineering, 2018-12-12:
[Realsense Customer Engineering Team Comment]
Ticket being closed due to inactivity for 30+ days

closed 2018-12-12 by RealSense-Customer-Engineering

Chris45215, 2021-07-28:
> Hello, pc=rs.pointcloud() pc.map_to(aligned_depth_frame) points = pc.calculate(aligned_depth_frame) pcl_points=pcl.PointCloud() point_to_pcl(pcl_points,points)

phojjat,
Sorry to revive a long-dead thread, but I've been looking for documentation on that point_to_pcl() function but I can't find it anywhere. In fact, if I put "point_to_pcl" in google (with the quotes), Google returns one result: this thread. Can you provide a bit more information on it, please?

Personally, I am using PCLpy as it seems to be better maintained than python-PCL, but the functionality between the two is very similar.  I can put a pointcloud in the PCLpy visualizer, but am having major issues when I try to texturize the pointcloud with the color from the RGB camera - the only way I found to do it is extremely inefficient and runs at about 1FPS.