realsenseai/librealsense #2484: Multiple Camera Calibration (D415) using opencv findHomography

metanamo, 2018-10-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

NguyenThaiHoc1, 2018-10-13:
Hi @metanamo 
I can a question for you, I hope you give answer  for my question 

How can i connect 3 camera in a computer and How can you display 3 camera in Intel-RealSense-Viewer app at the same time as you did 

Thanks you 
Thai Hoc,

metanamo, 2018-10-14:
>You need to have 3 different USB connections not using the same hub.
Then you can open Realsense Viewer and on the top left add all devices.

> 
> Hi @metanamo
> I can a question for you, I hope you give answer for my question
> 
> How can i connect 3 camera in a computer and How can you display 3 camera in Intel-RealSense-Viewer app at the same time as you did
> 
> Thanks you
> Thai Hoc,

RealSense-Customer-Engineering, 2018-10-23:
[Realsense Customer Engineering Team Comment]
Hi @metanamo,

The decomposeHomographyMat() should provide the rotation and translation.

https://docs.opencv.org/3.4.1/d9/d0c/group__calib3d.html#ga7f60bdff78833d1e3fd6d9d0fd538d92

Below contains an example to decompose the Homography matrix.
https://docs.opencv.org/3.4.1/d9/dab/tutorial_homography.html

Thank you.

closed 2018-12-12 by RealSense-Customer-Engineering