realsenseai/librealsense #2485: Failed to create ros reader: Error reading version line

philipjames44, 2018-10-08:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  SR300  | 
| Firmware Version                   | N/A | 
| Operating System & Version |    MacOS  | 
| Kernel Version (Linux Only)    |  N/A |                                         | 
| Platform                                 | PC |
| SDK Version                            |   2.16.0  | 
| Language                            | C++ }                          | 
| Segment			|  N/A  | 

### Issue Description
I've been getting an "Failed to create ros reader: Error reading version line" when trying to open an .rssdk file. My code is as so:

```
    #include <librealsense2/rs.hpp>
  4 
  5 using namespace std;
  6 using namespace cv;
  7 
  8 int main(){
  9 
 10 
 11 rs2::config cfg;
 12 cfg.enable_device_from_file("video2.rssdk");
 13 rs2::pipeline pipe;
 14 pipe.start(cfg); //File will be opened in read mode at this point
 15 for (int i = 0; i < 30; i++)
 16 {
 17     rs2::frameset frames;
 18     if (pipe.poll_for_frames(&frames))
 19     {
 20         //use frames here
 21     }
 22     //Note that we use poll_for_frames instead of wait_for_frames
 23     //This is because the file only contains a finite amount of frames
 24     // and if we use wait_for_frames then after reading the entire file
 25     // we will fail to wait for the next frame (and get an exception)
 26 }
 27 pipe.stop(); //File will be closed at this point
 28 
 29 }
```

And my full error message is:

`libc++abi.dylib: terminating with uncaught exception of type rs2::error: Failed to resolve request. Request to enable_device_from_file("video2.rssdk") was invalid, Reason: Failed to create ros reader: Error reading version line
Abort trap: 6`

dorodnic, 2018-10-08:
Hi @philipjames44 
Unfortunately, the `rssdk` files are only supported by now discontinued [SDK for Windows](https://software.intel.com/en-us/realsense-sdk-windows-eol).

philipjames44, 2018-10-08:
Ah, unfortunate, thanks for the response however. Is there a more common format/procedure I should record my videos in that is still supported?

RealSense-Customer-Engineering, 2018-10-08:
[Realsense Customer Engineering Team Comment]
Hi @philipjames44,

You can consider to record it as a ROS bag file.

philipjames44, 2018-10-09:
So a follow up question. Is there any way of opening/changing the rssdk format so that it can be read by this sdk (RealSense2)?

RealSense-Customer-Engineering, 2018-10-23:
[Realsense Customer Engineering Team Comment]
Hi philipjames44,

As dorodnic mentioned, rssdk is only supported by the discontinued SDK for Windows.

You can try either ROS and RealSense2 to record ROS bag file.
You may take a look at this example for recording and playing back: https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback

Thanks!

closed 2018-10-24 by philipjames44