realsenseai/librealsense #2486: Compilation Error: rs-align example on VS2017

HippoEug, 2018-10-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version | Win 10 | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC |
| SDK Version                            | 2.16.1 | 
| Language                            | C++  | 
| Segment			| others                   | 

### Issue Description
Hello once again,

Unfortunately I have ran into errors which I have spent 2 days trying to solve. I have asked this [question](https://github.com/IntelRealSense/librealsense/issues/2450) once before, and @dorodnic has given me guidance but I am still having trouble with it.

#### Things I have done
1. Created new empty C++ Project, copied [example.hpp, int-rs-splash.hpp, stb_easy_font.h, stb_image.h, stb_image_write.h] into File Explorer
2. Copied intel.realsense.props in Property Manager
3. Copied [imgui.cpp, imgui_draw.cpp, imgui_impl_glfw.cpp] into File Explorer
4. VS > Properties > C/C++ > General > Additional Include Dirs > Edit > (C:\Users\efoo1\Desktop\RealSenseCmake\third-party\imgui)
5. Created Source.cpp under source and copied the code from [rs-align.cpp](https://github.com/IntelRealSense/librealsense/blob/master/examples/align/rs-align.cpp) into it
6. VS > Solution Explorer > Source > Add Existing Item > and copied [imgui.cpp, imgui_draw.cpp, imgui_impl_glfw.cpp]

Here is how my Solution Explorer looks: 
[image]

#### Errors
1. Line 5 has errors, but I figured since I manually copied example.hpp into the project in File Explorer, #include "example.hpp" directly would make more sense.
2. Line 25 is a bigger error, it says "no suitable conversion function from "window" to "GLFWwindow*" exists".. I don't get it since this works on CMake.
[image]
3. If I comment Line 25 out, I still get a magnitude of errors.. on the imgui_impl_glfw.cpp file, not on Source.cpp.
[image]
4. I tried to add the GLFW library into the Properties Pages and shown below. (I downloaded the GLFW files and placed it on my Desktop)
[image]
4.1. Properties > VC++ > Include Directories > (C:\Users\efoo1\Desktop\glfw-3.2.1.bin.WIN64\include)
[image]
4.2 Properties > VC++ > Library Directories > (C:\Users\efoo1\Desktop\glfw-3.2.1.bin.WIN64\lib-vc2015)
[image]
4.3 Properties > Linker > Input > Additional Dependencies > glfw3.lib & glfw3dll.lib
[image]
5. Lastly, I copied the glfw3.dll into the Project File Explorer. 
[image]

#### Problems still
I am still getting so many errors during linking even after trying so many things...
[image]

#### CMake
NOTE: CMake works, but the problem is I cannot modify on it. Hence, I am trying to create the project from scratch and manually include the directories.. 

## I am losing my sanity.. any help would be 150% appreciated. Thank you guys, you're my last hope

HippoEug, 2018-10-08:
Edit 1: I just found out that on the CMake version of rs-align, the code inside #include "../example.hpp" is quite different from the #include "example.hpp" in the other simpler examples such as rs-align. So a simple copy edit of the #include is not possible...

Source Code:
[#include "../example.hpp"](https://hastebin.com/axugizawuy.cpp)
[#include "example.hpp"](https://hastebin.com/ewotijedak.cpp)

Will change it and see how it goes

RealSense-Customer-Engineering, 2018-10-09:
[Realsense Customer Engineering Team Comment]
Hi @HippoEug,

You can build the all librealsense project and open "RealsenseExamplesAlign.sln". Then, you can apply the similar settings, files and libraries to your rs-align project.

HippoEug, 2018-10-10:
Yeah I have tried that, had errors. But I think I found the fix for it.. but using 32bit binaries. Still gotta figure the x64 one out!

freemanlo, 2018-10-11:
My standalone rs-align project also attached for reference.
[rs-align-standaloneBuild.zip](https://github.com/IntelRealSense/librealsense/files/2467104/rs-align-standaloneBuild.zip)

HippoEug, 2018-10-12:
Thank you freemanlo, that's very helpful :D

RealSense-Customer-Engineering, 2018-10-22:
[Realsense Customer Engineering Team Comment]
Hi @HippoEug,

Still need any support for this topic?

dorodnic, 2018-10-22:
Note: in the upcoming release the installer is going to include rs-align, post-processing, playback-record and opencv examples based on @HippoEug and others feedback.

closed 2018-10-30 by RealSense-Customer-Engineering