realsenseai/librealsense #2487: Matlab wrapper: depth_sensor typos

izaakvc, 2018-10-08:
The Matlab wrapper contains some typos in both `realsense.depth_sensor` and `realsense.depth_stereo_sensor`:
line 1: `classdef depth_sensor < sensor` must be `classdef depth_sensor < `**`realsense.sensor`**
line 15: `emd` must be `end`

izaakvc, 2018-10-08:
And in `realsense.config`:
line 21:  `'<=', realsense.stream.count` should be `'<=', int64(realsense.stream.count)`
(to avoid the error: `Error using validateattributes Value for <= must be a real scalar.`)
line 22: semicolon missing at the end of the line.

line 86: `args = num2cell(int64([varargin{:}]));` causes conversion errors, for example: 
`config.enable_stream(realsense.stream.depth, realsense.format.z16)` results in `Error using realsense.stream Unsupported conversion from 'realsense.format' to enumeration 'realsense.stream'.`

This is solved when line 86 is replaced by: 
```
args = {size(varargin)};
for i = 1:length(varargin)
    args{i} = int64(varargin{i});
end
```

izaakvc, 2018-10-09:
In frameset.m (and the corresponding part in librealsense_mex.cpp) there are some small things missing:
`get_size` is not implemented in the mex file.
`get_infrared_frame`  has no index argument in the Matlab wrapper, while it does have it in rs_frame.hpp. While you can enable both left and right infrared streams, it is impossible to receive both images. You can only get the first infrared frame.

izaakvc, 2018-10-11:
In pipeline_profile.m:
line 30: `..., 'get_streams', ...` should be `..., 'get_stream', ...` (without the 's')
In stream_profile.m:
line 45: `realsense.stream_profile(out{:});` should be `profile = realsense.stream_profile(out{:});`

I noticed this when I was trying to get the intrinsics from the camera like [this](https://github.com/IntelRealSense/librealsense/wiki/API-How-To#get-video-stream-intrinsics).
When doing following steps:
```matlab
pipeline         = realsense.pipeline();
pipeline_profile = pipeline.start(obj.config);
depth_stream     = pipeline_profile.get_stream(realsense.stream.depth);
depth_stream     = depth_stream.as('video_stream_profile');      % <--- 'Access violation'
intrinsics       = depth_stream.get_intrinsics();
```
It results in a complete crash (Access violation detected) when executing the last line.
In fact, it crashes when accessing any of the functions of stream_profile (stream_name, stream_index, fps, ...)

izaakvc, 2018-10-11:
disparity_transform.m:
line 11: `end` missing

dorodnic, 2018-10-11:
Hi @izaakvc 
Yes, the recent post-processing refactoring is giving us troubles with regards to Matlab wrapping, but we are working to address it ASAP.

lramati, 2018-10-29:
With the exception of the comment moved to issue #2541 all these typos should now be fixed in development branch thanks to PR #2492

ev-mp, 2018-11-19:
Fixed by  [PR2644](https://github.com/IntelRealSense/librealsense/pull/2644)