realsenseai/librealsense #2494: Loading JSON Config Crashes Librealsense

xtrawurst, 2018-10-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D410 | 
| Firmware Version                   | 5.10.3 | 
| Operating System & Version |   Win10 | 
| Platform                                 | Panasonic FZ-M1 (integrated D410) |
| SDK Version                            |  2.16.1 (and prior) | 

### Issue Description
To reproduce:
(0. Start RS Viewer, select "Remove IR Pattern" config and save to JSON file. Quit RS Viewer.)
1. Start RS Viewer
2. Load the saved config file from step 0 or the one attached.
3. RS Viewer becomes unresponsive

The same thing happens when we load the JSON file in our app (using libRS) through the `rs400::advanced_mode::load_json()` function so it seems to be an issue in librealsense or the camera FW, not the RS Viewer.

[noir.json.txt](https://github.com/IntelRealSense/librealsense/files/2456452/noir.json.txt)

RealSense-Customer-Engineering, 2018-10-09:
[Realsense Customer Engineering Team Comment]
Hello @xtrawurst

Using D410 / FW 5.10.3 / Windows 10 RS4 / LibRealSense 2.16.1 and your steps:

(0. Start RS Viewer, select "Remove IR Pattern" config and save to JSON file. Quit RS Viewer.) = DONE
1. Start RS Viewer. = DONE
2. Load the saved config file from step 0 or the one attached. = DONE
3. RS Viewer becomes unresponsive. = An error occurs "Unsupported stream-ir-format: RGB8".  But press "OK" we are able to stream camera.  The attached .json you provide file does have the same results with error: "Unsupported stream-ir-format: RGB8" but again, we can interact with RS-Viewer after the pop-up dialog box shows.
A few questions:

-  When you say "RS Viewer becomes unresponsive".  Do you mean you cant interact with it all?  The window is frozen?  Or is the error only the dialog box that reports "Unsupported RGB8?"
-  Can you provide RealSense debug logs of issue?  https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A
(To change the log level of LibRealSense logger, you need to set a local variable named **LRS_LOG_LEVEL** and initialize it with the desirable log level:)
**Windows:**

    > set LRS_LOG_LEVEL="<Log Level>"<Log Level> can be either DEBUG, INFO, WARNING, ERROR or FATAL

-  Can you provide screenshots or video of crash?

Thank you!

RealSense-Customer-Engineering, 2018-10-10:
[Realsense Customer Engineering Team Comment]
Hi xtrawurst,

From the json file, it seems that you chose "RGB8" for Infrared1 stream when streaming with "Remove Ir Pattern" preset (this is fine). And then it was saved to the json file.

When loading the saved json file, RS Viwwer has a check on pixel format of IR stream and only allows "Y8" or "UYVY".
This is the reason that it failed on loading your saved json file.

If it is possible to your use case, as a workaround, you may change the IR format to "Y8" or "UYVY".
Or you may go ahead and patch up model-views.cpp file.
A similar case to the failure on calling load_json() API, but it uses a different check.

Hope this can help.

Thanks!

xtrawurst, 2018-10-10:
@RealSense-Customer-Engineering 
Thanks. Actually, RGB8 is what we need. In the end we'd like to have
* A depth stream at 1280x720
* An RGB8 color stream from the left IR camera with pattern removed.

We know this is possible but since recently get the crash.

> When you say "RS Viewer becomes unresponsive". Do you mean you cant interact with it all? The window is frozen?

Yes. The window freezes and we have to force-quit the RS Viewer through the task manager to get the window off the screen.
I will try to get you a screen recording of the freezing and the debug logs.

xtrawurst, 2018-10-10:
@RealSense-Customer-Engineering 
> Can you provide RealSense debug logs of issue?

I set the env. variable in the terminal and then started RS Viewer from the terminal. However I can't see any logs there. Please let me know where I can find the logs.

xtrawurst, 2018-10-10:
@RealSense-Customer-Engineering 
Here is a [video](https://www.dropbox.com/s/xt9fawkys796u9l/D410_RSViewer_freeze.mp4?dl=0) showing the error.

RealSense-Customer-Engineering, 2018-10-10:
[Realsense Customer Engineering Team Comment]
Hi xtrawurst,

Thanks for sharing the video and updates.
We can't reproduce the RS Viewer frozen issue here (Windows 10 PC, SDK 2.16.1, and 410 w/ FW 5.10.3). But only got an error msg about "Unsupported stream-ir-format: RGB8", which is caused by the RS Viewer check mentioned below (in model-views.cpp file).

As for enabling RS log, first you should open a Windows Cmd terminal from the directory where you have RS Viewer installed.
Inside the Cmd terminal, set the following env variable

    set LRS_LOG_LEVEL=DEBUG

And then run RS Viewer (with correct path info) from the Cmd terminal.
After finishing the test, you need to close the RS Viewer, and there will be a log file saved under the directory that you started RS Viewer.

On the other side, can you try same thing on a Windows 10 PC, and see what you will get?

Thanks!

xtrawurst, 2018-10-10:
@RealSense-Customer-Engineering 

> On the other side, can you try same thing on a Windows 10 PC, and see what you will get?

Unfortunately we don't have a D410 as a module. We only have one Panasonic FZ-M1 with an integrated D410.

RealSense-Customer-Engineering, 2018-10-10:
[Realsense Customer Engineering Team Comment]
Hi xtrawurst,

You can test w/ D415 if you have.
On the other side, please try to collect a log file as instructed, as we don't have the issue here.

Thanks!

RealSense-Customer-Engineering, 2018-10-10:
[Realsense Customer Engineering Team Comment]
Hi xtrawurst,
Can you help clarify a few more things:
1. For loading the json file using rs400::advanced_mode::load_json(), was your app frozen or what happened?
2. Were you able to do what you need before w/ other version of RS Viewer and/or FW versions?

Thanks!

xtrawurst, 2018-10-10:
> For loading the json file using rs400::advanced_mode::load_json(), was your app frozen or what happened?

Yes, the call to rs400::advanced_mode::load_json() never returned.

> Were you able to do what you need before w/ other version of RS Viewer and/or FW versions?

Yes. We did have a freezing issue before, however before I could get to the bottom of it I did a firmware upgrade (from 5.9.14 to 5.10.3) and that "fixed" it. However it only lasted some days then the issue described in this thread started to happen. So it did work with the 5.10.3 Firmware before and no change to librealsense. Not sure what else changed that caused the issue.

RealSense-Customer-Engineering, 2018-10-11:
[Realsense Customer Engineering Team Comment]
Hi xtrawurst,

Thanks for the update.
Can you please help get log file and also try the other tests suggested below?

Thanks!

RealSense-Customer-Engineering, 2018-10-11:
[Realsense Customer Engineering Team Comment]
Hi xtrawurst,

FYI
We managed to get a Panasonic FZ-M1 and configured to be with a D410 module (laoded with same SDK & FW as your case).
With several tests with same steps you have, we could not reproduce the issue with RS Viewer, but only saw "Unsupported stream-ir-format: RGB8" error message, and RS Viewer was still alive.

Thanks,
Richad

xtrawurst, 2018-10-12:
@RealSense-Customer-Engineering 

I was able to have the RS Viewer create a log file using `set LRS_LOG_LEVEL=DEBUG`, however the log file is empty.

RealSense-Customer-Engineering, 2018-10-12:
[Realsense Customer Engineering Team Comment]
For getting a log file, normally the contents will be dumped into a file after closing RS Viewer.
Not sure for your case, if it caused by RS Viewer is closed abnormally from Task Manager.

At this moment, we can't help on the frozen issue as we can't reproduce it.
Do you another set of the system to try?

Thanks!

RealSense-Customer-Engineering, 2018-10-22:
[Realsense Customer Engineering Team Comment]
Hi xtrawurst,

Any more update on getting a log file? Or do you still need any help?

Thanks!

xtrawurst, 2018-10-22:
@RealSense-Customer-Engineering 
For some reason I can no longer reproduce it either. I now also get the "Unsupported stream-ir-format: RGB8" error.

I didn't change anything in the configuration though. Could it have been a Windows update?

RealSense-Customer-Engineering, 2018-10-22:
[Realsense Customer Engineering Team Comment]
Hi xtrawurst,

Thanks for the update.
As for your current "Unsupported stream-ir-format: RGB8" error, please refer to what have been mentioned.

Thanks!

RealSense-Customer-Engineering, 2018-10-29:
[Realsense Customer Engineering Team Comment]
Hi xtrawurst,

We are going to close this one if there is nothing else.

Thanks!

closed 2018-10-30 by RealSense-Customer-Engineering