realsenseai/librealsense #2495: Strange Depth Playback Issues

philipjames44, 2018-10-08:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Operating System & Version |   MacOS  | 
| Platform                                 | PC  |
| SDK Version                            |  2.16.0                          | 
| Language                            | C++             | 

### Issue Description
I've been getting very strange depth data when displaying it using the code below. It's almost as though the depth image gets split into 3 overlapping channels? Color video plays back fine, but depth never works.

Image of sample output:
[image]

Code:
```
#include "opencv2/opencv.hpp"
  2 #include <iostream>
  3 #include <librealsense2/rs.hpp>
  4 using namespace std;
  5 using namespace cv;
  6 
  7 int main(){
  8 
  9 using namespace cv;
 10 const auto window_name = "Display Image";
 11 namedWindow(window_name, WINDOW_AUTOSIZE);
 12 
 13 rs2::config cfg;
 14 cfg.enable_device_from_file("/Users/phil/Documents/20181008_182015.bag");
 15 rs2::pipeline pipe;
 16 pipe.start(cfg); //File will be opened in read mode at this point
 17 
 18 
 19 int num = 0;
 20 while(num != 100 && waitKey(1)){
 21         num++;
 22         rs2::frameset data;
 23         pipe.poll_for_frames(&data);
 24         rs2::frame depth = data.get_depth_frame();
 25         if(!depth){
 26                 cout << "nah" << endl;
 27         }
 28         else{
 29         // Query frame size (width and height)
 30         const int w = depth.as<rs2::video_frame>().get_width();
 31         const int h = depth.as<rs2::video_frame>().get_height();
 32         // Create OpenCV matrix of size (w,h) from the colorized depth data
 33         Mat image(Size(w, h), CV_8UC3, (void*)depth.get_data(), Mat::AUTO_STEP);
 34         // Update the window with new data
 35         imshow(window_name, image);
 36         waitKey(0);
 37         }
 38 }
 39 
 40 pipe.stop();
 41 return EXIT_SUCCESS;
      }

```

dorodnic, 2018-10-09:
Hi @philipjames44 
Probably because of `CV_8UC3`.
In [im-show](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/imshow/rs-imshow.cpp) we are first converting depth data to an RGB image via `rs2::frame depth = data.get_depth_frame().apply_filter(color_map);`

philipjames44, 2018-10-09:
Ah yes this fixed it. Thanks for your patience (as it was indeed in the documentation) :)

closed 2018-10-09 by philipjames44