realsenseai/librealsense #2497: USB2.1 infra-red exposure issue for short exposure times (<70usec)

m0oz, 2018-10-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.10.03.00 | 
| Operating System & Version |   Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    |  4.15.0 | 
| Platform                                 | PC |
| SDK Version                            |  2.16.0  | 
| Language                            |  C++   | 
| Segment			|  Robot  | 

## Issue Description
When connected to a USB2.1 port one cannot set the depth exposure time shorter than 70usec.
Exp times shorter than 70usec will lead to artifacts and over-exposed frames.

## Steps to Reproduce
* Connect a D435 to a USB 2.1 port and open realsense-viewer
* Enable infra1 stream and set "enable auto exposure" to OFF
* Set exposure time to something like 100usec and you get unaffected frames
* Set exposure time to any value lower than 70usec and the frame becomes over-exposed (it looks brighter than at 100usec) and will show artifacts.

ATTENTION: When using a USB3.2 port we could not observe this issue

| *USB2.1, 90usec exposure* | *USB3.2, 90usec exposure* |
|---------------------------------|------------------------------------------- |
| [image]| [image]| 

| *USB2.1, 60usec exposure* | *USB3.2, 60usec exposure* |
|---------------------------------|------------------------------------------- |
|[image] | [image]|

### USB2.1, 1usec exposure, artifacts (salt and pepper)
[image]

RealSense-Customer-Engineering, 2018-10-11:
[Realsense Customer Engineering Team Comment]
Hello @m0oz,

Thank you for pointing out this issue!  We were able to replicate the issue with RealSense Viewer and will work to create a fix shortly.
Please stay tuned, will update this issue when fix is implemented.

mhkabir, 2018-12-06:
@RealSense-Customer-Engineering Any updates?