realsenseai/librealsense #2498: librealsense2-dkms package broken on Ubuntu 16.04.5 LTS

phil-marble, 2018-10-09:
I think this is related to #2402 
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Ubuntu 16.04.5 LTS  | 
| Kernel Version (Linux Only)    |           4.15.0-36-generic #39~16.04.1-Ubuntu                             | 
| Platform                                 | PC |
| SDK Version                            |           2.12.0-0~realsense0.73                | 
| Language                            |  N/A                       | 
| Segment			|                          | 

### Issue Description

a newer `librealsense2-dkms` package is not currently available from your repo that claims to support xenial:

```
phil@phil-870-120:~$ apt-cache showpkg librealsense2-dkms
Package: librealsense2-dkms
Versions: 
1.2.0-0ubuntu5 (/var/lib/dpkg/status)
 Description Language: 
                 File: /var/lib/dpkg/status
                  MD5: 2a019f58c43f19f75d77c098047b9dcd

Reverse Depends: 
Dependencies: 
1.2.0-0ubuntu5 - dkms (2 2.1.0.0) librealsense2-udev-rules (0 (null)) 
Provides: 
1.2.0-0ubuntu5 - 
Reverse Provides:
```

``` 
phil@phil-870-120:~$ cat /etc/apt/sources.list.d/realsense.list 
deb "http://realsense-alm-public.s3.amazonaws.com/apt-repo" xenial main
```

Earlier Communication:

```
Dkms package version 1.2 is limited to kernel 4.13 and is not up to date.
It has been superseded by couple of versions, iirc 1.3.1 is the latest ,which definitely supports kernel 4.15.

Please switch to the appropriate forum https://github.com/IntelRealSense/librealsense/issues for further assistance.

BR,
Evgeni

-----Original Message-----
From: Phil Freeman 
Sent: Wednesday, October 03, 2018 20:19
To: Raikhel, Evgeni 
Subject: librealsense2-dkms package broken on Ubuntu 16.04.5 LTS

Hello,

 It looks like this package is failing on Ubuntu 16.04.5 LTS with kernel 4.15.0-36-generic #39~16.04.1-Ubuntu

phil@phil-870-120:~$ sudo apt-get install librealsense2-dkms librealsense2-utils Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  librealsense2-dkms librealsense2-utils
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 5,322 kB of archives.
After this operation, 38.1 MB of additional disk space will be used.
<SNIP>
Selecting previously unselected package librealsense2-dkms.
(Reading database ... 402434 files and directories currently installed.) Preparing to unpack .../librealsense2-dkms_1.2.0-0ubuntu5_all.deb ...
Unpacking librealsense2-dkms (1.2.0-0ubuntu5) ...
Selecting previously unselected package librealsense2-utils:amd64.
Preparing to unpack
.../librealsense2-utils_2.12.0-0~realsense0.73_amd64.deb ...
Unpacking librealsense2-utils:amd64 (2.12.0-0~realsense0.73) ...
Setting up librealsense2-dkms (1.2.0-0ubuntu5) ...
Loading new librealsense2-dkms-1.2.0 DKMS files...
First Installation: checking all kernels...
Building only for 4.15.0-36-generic
Building initial module for 4.15.0-36-generic ERROR (dkms apport): kernel package linux-headers-4.15.0-36-generic is not supported Error! Bad return status for module build on kernel: 4.15.0-36-generic
(x86_64)
Consult /var/lib/dkms/librealsense2-dkms/1.2.0/build/make.log for more information.

Loading the modified modules into kernel... complete Current status:
<SNIP>
librealsense2-dkms, 1.2.0: added
<SNIP>
mod: videodev
mod: uvcvideo                    version: 1.1.1
mod: hid_sensor_gyro_3d
mod: hid_sensor_accel_3d
Setting up librealsense2-utils:amd64 (2.12.0-0~realsense0.73) ...
```

RealSense-Customer-Engineering, 2018-10-09:
[Realsense Customer Engineering Team Comment]
Hello @phil-marble

We installed a new image of Ubuntu 16.04.5; followed these instructions and were not able to replicate the issue you are seeing.

Librealsense-dkms version 1.3.1 is available on AWS right now, and works with Ubuntu 16.04.5 Kernel 4.15.0-36.  (See attached screenshots)

Suggested Steps:
>  Remove a single package with:  (Replace <package-name> with "librealsense-dkms")
`sudo apt-get --purge <package-name>`

(OR) Remove all RealSense™ SDK-related packages with:
 `dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

>  Remove the old records:

-

`sudo rm -f /etc/apt/sources.list.d/realsense-public.list`.

-

`sudo apt-get update`.

>  Follow these instructions again.

Please let us know if you are able to retrieve package "librealsense-dkms" version 1.3.1.

Thank you!

phil-marble, 2018-10-09:
Can you provide a link to what you are referring to here as "these instructions"?

RealSense-Customer-Engineering, 2018-10-09:
[Realsense Customer Engineering Team Comment]
@phil-marble

My apologies, here is the link to the instructions:  https://github.com/IntelRealSense/librealsense/blob/v2.16.1/doc/distribution_linux.md

Thank you!

phil-marble, 2018-10-09:
Ah, I think I had the wrong URL configures... Thanks!

closed 2018-10-09 by phil-marble