realsenseai/librealsense #2499: pyrealsense2: 'pc_calculate' only returns part of point cloud

LLDavid, 2018-10-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | 05.10.03.00 | 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |            -                           | 
| Platform                                 | PC |
| SDK Version                            | -                      | 
| Language                            | Python }                          | 
| Segment			| -                      | 

Hi, 
I am trying to stream point cloud from D415 using code in the example:

```
    while True:
        # Wait for the next set of frames from the camera
        frames = pipeline_1.wait_for_frames()
        # Fetch color and depth frames
        depth = frames.get_depth_frame()
        color = frames.get_color_frame()
        if not depth or not color:
            continue
        frame_id=frame_id+1
        # Tell pointcloud object to map to this color frame
        pc.map_to(color)
        # Generate the pointcloud and texture mappings
        points = pc.calculate(depth)
        vtx = np.asarray(points.get_vertices())
        # Save to .ply
        print("Saving to i.ply...")
        file_path='.\\ply\\'+str(frame_id)+'.ply'
        points.export_to_ply(file_path, color)
        print("Done")
        # Read PLY
        pcd = read_point_cloud(file_path)
        # Visualize PLY
        draw_geometries([pcd])
```
However the output is quite weird with  point cloud of only a few rays (I put the cam toward my face, and I tested with the alignment example, and the cam is fine). It seems like lots of point cloud are missing:
[image]
I already posted the question in the forum [(link)](https://communities.intel.com/thread/129663).
Does any one know how to deal with it? Thanks in advance

dorodnic, 2018-10-09:
Hi @LLDavid 
Did you try to open the resulting PLY in some other software like MeshLab?

LLDavid, 2018-10-09:
> Hi @LLDavid
> Did you try to open the resulting PLY in some other software like MeshLab?

No. But I tried using the matplotlib to plot the 3D points directly.
[image]
And I get similar result. So I think it is probably not the viewer's problem.

LLDavid, 2018-10-11:
> Just cut off depth measurements over a threshold. Notice that you have distance measurements at 17.5m, which is quite big distance, so your viewer tries to normalize it.

@vladsterz  Thanks! It sounds like the reason. But do you know how to set filter to the depth image in python?  Since the the 'depth' is a class, the tutorial example only shows how to filter the depth after transform the depth class to array

vladsterz, 2018-10-11:
@LLDavid The most straightforward way to do it is to set the depth value of a pixel to zero, if it exceeds a threshold. Unfortunately I am not familiar with the Python API, so I cannot give you any insights on this. But there is a more elegant way of achieving your goal, set the depth threshold in the depth sensor. Once again, not familiar with the API, so no much I can do except [this](https://github.com/IntelRealSense/librealsense/issues/2199#issuecomment-410683830).

LLDavid, 2018-10-11:
> @LLDavid The most straightforward way to do it is to set the depth value of a pixel to zero, if it exceeds a threshold. Unfortunately I am not familiar with the Python API, so I cannot give you any [...]

@vladsterz The reason I stick to python is beacause I want to apply some  DNN code to the point cloud. But as you said, the C++ API seems have more flexibility. I will probably do the intrinsic calibration my own code if there is no straight solution to this. Thanks for the comment!

closed 2018-10-15 by LLDavid