realsenseai/librealsense #2500: Error when building pyrealsense2 on Raspberry Pi

eospi, 2018-10-09:
_Originally posted by @eospi in https://github.com/IntelRealSense/librealsense/issues/2496#issuecomment-428193008_

I've successfully installed the SDK on my Raspberry Pi, but when I went to build the Python library, the `make -j4` fails. The error message says `recipe for target 'all' failed`.

dorodnic, 2018-10-10:
Hi @eospi 

There are couple of things you need to check:

1. You have a swap file set up (otherwise gcc is very likely to crash because of out of memory)
2. Verify you have GCC 5
3. Do not add `-j4` - on a Raspberry Pi trying to compile on more then one core is a bad idea.

RealSense-Customer-Engineering, 2018-10-10:
[Realsense Customer Engineering Team Comment]
Hello @eospi

We were able to replicate what you are seeing, it indeed has to do with swap file space, or lack thereof.

After following these steps the python bindings were able to be compiled:

-  Enable swap file, for our tests we enabled 1GB:  https://www.digitalocean.com/community/tutorials/how-to-add-swap-space-on-ubuntu-16-04

-  Verify you are compiling from main SDK directory, not the python folder.  https://github.com/IntelRealSense/librealsense/issues/1220
-  If you are still experiencing issues compiling wrappers, disable OpenGL - if you have already enabled it, I have noticed OpenGL being enabled uses more environment resources than when it is turned off.
     Steps: - sudo raspi-config

- 5 Advanced Options

- A6 GL Driver

-  G3 Legacy Original non-GL desktop driver (Enable this if you still can not compile after following above steps.  Remember to enable G1 GL (Full KMS) OpenGL driver setting after wrapper compilation to allow for OpenGL to run camera streams.)

Please let us know if you have any other questions.
Thank you!

eospi, 2018-10-10:
Thanks for the help! After trying the swap file, it successfully compiled. I'm excited to get started on my project.

closed 2018-10-10 by eospi