realsenseai/librealsense #2504: Displaying Camera Frames in Custom OpenGL Window?

HippoEug, 2018-10-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version | Win 10 | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC |
| SDK Version                            | 2.16.1 | 
| Language                            | C++  | 
| Segment			| others                   | 

### Issue Description
Good morning/afternoon/evening,

I am trying to create a custom version of Realsense Viewer with additional function using the D435. Here is a rough sketch up of the output currently using OpenGL and Dear ImGui.
[image]

Here is what I am trying to do:
[image]
[image]

As simple as this! Displaying the streams onto my own OpenGL Window using C++.

Here is my current code: 
glfwInit();
GLFWwindow* window = glfwCreateWindow(1280, 720, "IntelRealsenseX", NULL, NULL);
glfwMakeContextCurrent(window);
ImGui_ImplGlfw_InitForOpenGL(window, true);

Here is the example code from realsense SDK:
window app(1280, 720, "CPP - Align Example");
ImGui_ImplGfw_Init(app, false);
rs2::colorizer color_map;
texture renderer;
...
renderer.render(other_frame, altered_other_frame_rect); // I believe this is the output to OpenGL Window?

Another simpler example from realsense SDK:
window app(1280, 600, "rs-capture");
...
rs2::frame depth = color_map(data.get_depth_frame());
rs2::frame color = data.get_color_frame();
depth_image.render(depth, {0, 0, app.width()/2, app.height() });
color_image.render(color, { app.width()/2, app.width()/2, app.height() });

How do I replicate this but on my own GLFWwindow* window declaration? I am finding it difficult to understand how to render these rs2::frames.. 

Thank you for helping!

dorodnic, 2018-10-10:
Hi @HippoEug 
Have you looked at [example.hpp](https://github.com/IntelRealSense/librealsense/blob/master/examples/example.hpp#L59-L150) implementation? 
Once you have your GLFW window set up, you need to create OpenGL texture object, bind it and upload pixel values to it on every camera frame using `glTexImage2D`. Then on every application frame, you want to render 2D quad with this texture (we are using `glBegin / glEnd` but you can also do this with vertex buffers and other ways)

HippoEug, 2018-10-11:
Hi Dorodnic, I am stuck on a even simpler step before what you said. My issue is more of a general C++ programming one + OpenGL rather than the Realsense SDK issue. Have never used OpenGL, so am not familiar with it

[image]
Here is what I am trying to do. Create the GLFW window in say a .cpp file on its own, main.cpp in this case. Upon calling it, I want to pass this GLFW window object to another .cpp file for rendering.. is this possible? If it is not a easy solution, I think I am going to put the Realsense Libraries and the OpenGL/GLFW libraries together in a .cpp file and do it from there. (ie making the GLFW window in a .cpp file and then rendering it on the same file without passing anything).

NOTE: in the screenshot you see, _texture_s_ and _render_s_ is a exact copy of _texture_ and _render_ from example.hpp

Thank you so so much for your help :)

dorodnic, 2018-10-11:
With GLFW you can't really access the content of objects.
To get width and height of the window, you just call `glfwGetWindowSize(window, &_width, &_height);`
Please take a look at [cpp-capture.cpp](https://github.com/IntelRealSense/librealsense/blob/legacy/examples/cpp-capture.cpp). The calls to RealSense API are slightly different, but it puts much more emphasis on inlining all the OpenGL stuff into a single tutorial.

HippoEug, 2018-10-12:
> With GLFW you can't really access the content of objects.
> To get width and height of the window, you just call `glfwGetWindowSize(window, &_width, &_height);`
> Please take a look at [old cpp-capture.cpp](https://github.com/IntelRealSense/librealsense/blob/legacy/examples/cpp-capture.cpp). The calls to RealSense API are slightly different, but it puts much more emphasis on inlining all the OpenGL stuff into a single tutorial.

Hi Dorodnic, the example is pretty helpful. Thank you very much and have a great weekend!

HippoEug, 2018-10-16:
[image]

Yay! I somehow managed to do it, still need lots of work on it though..

RealSense-Customer-Engineering, 2018-10-23:
[Realsense Customer Engineering Team Comment]
Hi HippoEug,

Do you still need help on this or we can close it?

Thanks!

HippoEug, 2018-10-23:
You can close this, thanks!

closed 2018-10-23 by RealSense-Customer-Engineering