realsenseai/librealsense #2506: Driver Installation D435 on Windows 7

navona, 2018-10-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version |   Win 7, 64-bit  | 
| Platform                                 | PC  |
| SDK Version                            | Intel® RealSense™ SDK 2.0 (build 2.16.1)   | 
| Language                            |  opencv/pcl/python  | 

### Issue Description
I am trying to use my Intel RealSense D435 on a Windows 7 PC. Thank you for adding Win7 support in the latest 2.16.1 SDK release. Unfortunately I have problems to install the driver.

I installed the latest SDK on a Windows 7, 64-bit PC. After installation the camera did not work immediately. But "Search automatically for driver software" from device manager helped and I could connect to the camera modules from RealSense Viewer.

By now I would like to install the drivers on a different PC (Win7, 64-bit) which has no internet connection. I did the same as before but searching automatically for driver software did not work.
I tried to install the drivers manually (Selecting IntelRealSense_D400_series_win7.inf) from Device Manager. This worked for one module. For the other module I get an error when selecting the .inf file:

[image]

Do you have any ideas what the problem is?

I noticed that on the second (problematic) PC the hardware IDs are USB\VID_8086&PID_0AD6&MI_00 and
USB\VID_8086&PID_0AD6&MI_03 while on other PCs I get USB\VID_8086&PID_0B07&MI_00 and USB\VID_8086&PID_0B07&MI_03 for the same camera.

My device manger now looks like this:
[image]

It says for Intel(R) RealSense(TM) 430 with RGB Module Depth : This device cannot start. (Code 10)

Thanks in advance!

dorodnic, 2018-10-10:
Hi @navona 
Thank you for the feedback. If you install the [Intel.RealSense.SDK.Win7.exe](https://github.com/IntelRealSense/librealsense/releases/download/v2.16.1/Intel.RealSense.SDK.Win7.exe) while you have the camera connected, the installer should prompt you about setting up the driver. 
Installing manually using the INF is also an option, but I would avoid "Search automatically for driver software" and always go for the specific path. 
`0B07` is the right PID for the D435, and it enumerates as `0AD6` when connected to USB2. This should be Ok for driver installation (I think?), but you can try changing the port.

navona, 2018-10-11:
The problem was that my USB 3.0 ports worked as USB 2 ports. I had to install Intel(R) USB 3.0 Host Controller Switch Driver (from https://downloadcenter.intel.com/product/65855/Intel-USB-3-0-eXtensible-Host-Controller-Driver) which made my USB 3.0 ports now really work as USB 3.0.
The driver installation for Intel RealSense D435 worked then perfectly.
Thanks for your help.

closed 2018-10-11 by navona