realsenseai/librealsense #2507: Converting the integer array data to openCV Mat

RohanKPatil, 2018-10-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | RealSense viewer 2.16.0 | 
| Operating System & Version |   Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  4.15.0-33-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            |  2.0                         | 
| Language                            |  C++                           | 
| Segment			|  Build                         | 

### Issue Description
I am trying to generate `cv::Mat` using the `uint8_t` array obtained from `get_frame_data()` for a frame. I am building a client-server application where I am sending the integer array from server to client. At the client side, I am trying to generate `cv::Mat` from the integer array received. But, the image displayed is some garbage. I know `Mat()` also takes the `void *` array but I can't send this array from server to client, I need to convert it into `int `array first. So, how do I use the `int `array to convert it into Mat?
I have written the following code. Please help!

```
rs2::frame color_frame = frames.get_color_frame();
    int height = 480;
      int width = 640;
      int arr[width*height];
      auto pixels = (uint8_t*)color_frame.get_data();
      for(auto i=0;i<(width*height);i++)
      {
      	arr[i] = pixels[i];
      }

    // Creating OpenCV Matrix from a color image
    Mat color(Size(640, 480), CV_8UC3, arr, Mat::AUTO_STEP);

    // Display in a GUI
    namedWindow("Display Image", WINDOW_AUTOSIZE );
    imshow("Display Image", color);
```

dorodnic, 2018-10-10:
Hi @RohanKPatil 
I don't fully understand why you are trying to convert the data to `int`s? This way, every pixel will get replaced by 12-bytes instead of 3 for R, G and B. If you must have an array of integers, just using `memcpy` would probably do the trick.

RohanKPatil, 2018-10-10:
Sorry for the vague description. What is the other way that you suggest to send the frame data from server to the client? I am currently sending the array obtained by `get_frame_data()` . As this returns an array with `(void *)` to it, so I thought of converting it into int array first and then send the array. How can I use `memcpy`? Can you elaborate a bit?

RealSense-Customer-Engineering, 2018-10-11:
[Realsense Customer Engineering Team Comment]
From server, you can still send uint8_t array to client.
From client, obtain the uint8_t arrary and convert cv::Mat as below example code for color image.

  Mat colorMat(Size(width, height), CV_8UC3, (void*)pixels, cv::Mat::AUTO_STEP);
  imshow("color", colorMat);

RohanKPatil, 2018-10-11:
I have done the same thing. But the issue is that I am getting some garbage image. Probably the pixel values are getting disturbed during this int conversion. What is the alternate way to do this?

closed 2018-12-12 by RealSense-Customer-Engineering