realsenseai/librealsense #2508: Initialisation is very slow under MacOS Mojave

cansik, 2018-10-10:
After the updating from Sierra to Mojave, the Realsense D435 start up is really slow (30 seconds or more). It does not matter which framework (2.2.15 or 2.2.16.1) and it does not matter which wrapper I use (C++, C#, Java).

With `startup` I mean the time till the first frame is delivered. Every call to the library is slow, which is very strange.

Does anyone else also have this problem?

dorodnic, 2018-10-10:
Interesting. Thanks for reporting this, we will try to reproduce and see what's up.
Any additional information you can share (what Mac device are you using?)

cansik, 2018-10-14:
@dorodnic Ok the problem was really just something on my MacBook. Did a clean install today and now everything runs smooth again. I think Mojave messed something up during the update process.

closed 2018-10-14 by cansik