realsenseai/librealsense #2509: fatal error: librealsense2/rs.hpp: No such file or directory

JSully11, 2018-10-11:
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | 05.09.14.00 | 
| Operating System & Version |   Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  4.15.0-36-generic   | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.16.1}      | 
| Language                            |  {C++/opencv/ }    | 
| Segment			|  others    | 

### Issue Description
I am new to intel realsense sdk and trying to learn how to use it within my project. To start I have been trying to learn how the examples provided on here for OpenCV work. I tried to run the "rs-imshow.cpp" example located here: https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/imshow/rs-imshow.cpp

but I get this error immediately:

/home/joshua/realsense/imshow/imshow.cpp:1:10: fatal error: librealsense2/rs.hpp: No such file or directory
 #include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/imshow.dir/build.make:62: recipe for target 'CMakeFiles/imshow.dir/imshow.cpp.o' failed
make[2]: *** [CMakeFiles/imshow.dir/imshow.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/imshow.dir/all' failed
make[1]: *** [CMakeFiles/imshow.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

How do i go about fixing this? Thanks for your help in advance.

RealSense-Customer-Engineering, 2018-10-11:
[Realsense Customer Engineering Team Comment]
Hi @JSully11,

Please refer to below info about the build process:
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv

JSully11, 2018-10-11:
I am confused as what step four is telling me to do? I have OpenCV installed, I have run the second command, and i have built librealsense. But where would i add this command " -DBUILD_CV_EXAMPLES=true" to your cmake command. I get that it says cmake command, but what does that mean exactly. This is my first ever project using C++ and I'm just trying to learn/understand it all.

HippoEug, 2018-10-11:
> I am confused as what step four is telling me to do? I have OpenCV installed, I have run the second command, and i have built librealsense. But where would i add this command " [...]

Hi Joshua, I started on this project 5 weeks ago, and asked the same question as you exactly, first on Intel's own forums. I am pretty new on C++ too, and have never even heard of CMake beforehand. Anyway, after weeks of trying to figure out, I wrote my own guide, without using CMake. I hope this helps you...

[Getting #include <librealsense2/rs.hpp> to work](https://github.com/HippoEug/IntelRealsenseX/blob/master/README.md). I think the most important thing is trying to get rs-example.cpp to work first. If you can, it means you have successfully used Intel Realsense SDK, only then try to combine it with OpenCV, at least that is how I did it. 

If you need any screenshot or further clarifications just send it here. I can feel your pain haha!

JSully11, 2018-10-11:
@HippoEug Your instructions seem to be for a windows operating system, is that correct? I am using a linux ubuntu 18.04 operating system and your guide does not seem to align with the steps i need to do this.

JSully11, 2018-10-11:
> Hi @JSully11
> Could you please share the exact list of terminal commands?
> The [installation instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) have too many twists and turns, but I'm confident that it will be Ok.
> The most likely fix is IMHO switching the current directory to `/home/joshua/realsense/build` when running `cmake` and `make` commands.
@dorodnic 
The commands I used when building the sdk are as follows in order:

1. sudo apt-key adv --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key C8B3A55A6F3EFCDE

2. sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u

3. sudo rm -f /etc/apt/sources.list.d/realsense-public.list

4. sudo apt-get update

5. sudo apt-get install librealsense2-dkms

6. sudo apt-get install librealsense2-utils

when you say swithcing the current directory to my /home/joshua/realsense/build folder. How do i go about doing that, or where would i do that if i were to uninstall the libraries and restart from scratch. The steps up to that point would be very helpful.

Thanks, Josh

dorodnic, 2018-10-11:
Oh, I see.
All opencv instructions and examples currently assume they are being built from source inside the library, and examples CMake does not play well with installed version of the library. This is something we need to address.
That said, if you uninstall `librealsense2-utils` and follow the regular instructions for building from source, everything will work. You can safely skip all the kernel patching and udev rules parts because `librealsense2-dkms` is already installed.

HippoEug, 2018-10-12:
My bad, did not see your OS. Hopefully you managed to get your the SDK to work!

JSully11, 2018-10-12:
@dorodnic So are you saying i just need to uninstall that one package? Or do i need to start with a fresh build altogether? If i just needed to uninstall that package then it did not work for me. I still got the same error. I found something that said that opencv 4.0 won't work with some aspects of librealsense and suggested uninstalling it and reinstalling opencv 3.4.3, which i have now done, but i continue to get similar errors when running the examples.

dorodnic, 2018-10-12:
After uninstalling the package:
```sh
git clone https://github.com/IntelRealSense/librealsense.git
cd librealsense
mkdir build && cd build
cmake .. -DBUILD_CV_EXAMPLES=true
make -j4
```

JSully11, 2018-10-12:
I have it working for the time being now. I just wanted to upload my solution to this string incase anyone else has the issue. So first, I completely uninstalled both opencv 4.0 and Librealsense from my machine. Then i installed OpenCV 3.4.3 using the following:
[OpenCV Installation in Linux_.pdf](https://github.com/IntelRealSense/librealsense/files/2474028/OpenCV.Installation.in.Linux_.pdf)

Then i reinstalled the Realsense Sdk from the source using the commands in Bold
[Realsense Installation.pdf](https://github.com/IntelRealSense/librealsense/files/2474040/Realsense.Installation.pdf)

After this i was able to run the Opencv and realsense examples that were located in /usr/local/bin.

I hope this helps anyone that needs it. And a special thanks to @dorodnic for all your help and suggestions! They definitely lead me down the right path!

Josh

closed 2018-10-22 by RealSense-Customer-Engineering

honghande, 2019-01-09:
Hi，@dorodnic ，I have installed Intel RealSense SDK 2.0 and added intel. realsense. props to the property manager of vs2015：Unable to open source file "librealsense/rs.hpp"

HippoEug, 2019-01-10:
@honghande , just a suggestion that might be worth a shot.

Instead of writing `#include "librealsense2\rs.hpp"` write `#include <librealsense2\rs.hpp>`.

jacksonkr, 2019-01-30:
@HippoEug in my scenario the cpp was already using angle brackets instead of quotes

st2yang, 2021-07-18:
I met this issue on a new machine, and it turns out that I forgot to install opencv3 first before installing librealsense.