realsenseai/librealsense #2510: Record issue

YSH6, 2018-10-11:
### Issue Description
<Describe your issue / question / feature request / etc..>
[image]

When I recorded stream,it showed that
Unknown in rs2_create_record_device(device:00000282D05760E0, file:C):
Error opening file: C:\Users\\20181011_161059.bag

How to fix it?
Thanks

HippoEug, 2018-10-11:
I think it would be helpful to share the Realsense Viewer version/SDK version and your camera firmware version :)

dorodnic, 2018-10-11:
Hi @YSH6 
In the Viewer the error seem to state "Error opening file: C:\Users \ \ 20181011_161059.bag" (with double forward slashes). Is it possible your username is blank or has some special characters? Did you try to run the app as administrator? The Viewer is trying to write to your user's home directory

YSH6, 2018-10-11:
> I think it would be helpful to share the Realsense Viewer version/SDK version and your camera firmware version :)

Thank you,I just ran the app as administrator then the problem was solved

YSH6, 2018-10-11:
> Hi @YSH6
> In the Viewer the error seem to state "Error opening file: C:\Users \ \ 20181011_161059.bag" (with double forward slashes). Is it possible your username is blank or has some special [...]

It works,thank you !

closed 2018-10-12 by RealSense-Customer-Engineering