realsenseai/librealsense #2512: Direct control the external pins

Jiloc, 2018-10-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 

### Issue Description
Is it possible to directly control the external pins of the D400 devices? I can't find any example of it. I am interested to know if is possible to directly control the power pin to switch the external device on and off

dorodnic, 2018-10-11:
Hi @Jiloc 
I'm fairly certain this is impossible. You can use `RS2_OPTION_INTER_CAM_SYNC_MODE` to get signal out of the device every time it takes a frame, but these are not general purpose pins.

MartyG-RealSense, 2018-10-11:
Over at the Intel Support site, users who tried getting a signal from Pin 1 early in 2018 before hardware sync support in firmware was released said it was too noisy to be usable.  One user reported:

"We tried to read out pin 1 from the external sensors sync connector (GVSYNC0) while enabling the "Output Trigger" option in the RealSense Viewer but we could only measure a very noisy signal containing shutter noise, and not a clear trigger signal (e.g. 5V TTL)"

The main discussion about the pins at Intel Support can be found at the link below.

https://communities.intel.com/message/558842#558842

Jiloc, 2018-10-12:
Thank you both for the answer, everything is cleare now.

Just FYI (or the interested of someone that will look for it in future) I tested that the pins are powered even if the camera is just attached to the computer and not only when the camera is turned on by the software.

closed 2018-10-12 by Jiloc