realsenseai/librealsense #2514: Sporadic error when starting camera

BGA3, 2018-10-11:
Hi

I am using the D415 on a arm64 ubuntu (16.04) platform. It usually runs fine, but it sporadically stops, and my c++ programs tries to recover by restarting (both the OS, and subsequently by calling rs2::hardware_reset();. This is a big issue for my application, as the solution is to manually restart the setup.
I get the following errormessage when trying to start the camera:
RealSense error calling rs2_pipeline_start_with_config(pipe:0x208cb5d0, config:x208b5be0): set_xu(ctrl=1) failed! Last Error: Input/output error

I am running camera-firmware version 5_10_3_0, and API version 2.16.1.

dorodnic, 2018-10-11:
Hi @BGA3 
What platform are you running on? (ODroid, RockChip, Jetson, something else)

BGA3, 2018-10-11:
Its running on a Firefly ROC-RK3328-CC board (arm64 A53 cpu).

dorodnic, 2018-10-14:
Hi @BGA3 
I managed to get my hands on this board. When connected to one of the USB2 ports the camera seems to behave pretty well, the USB3 port, however, is giving me some troubles. 
We finally managed to make it to work by connecting the camera via powered USB3 hub. I'll let it run for some time. The errors I encountered are not "Input/Output error" but might share the same root cause. 
Could you please share `dmesg` right after encountering the problem? In my case, `dmesg` log was indicating a disconnect-reconnect (hinting at lack of power)

BGA3, 2018-10-16:
Hi @dorodnic 

Sounds great that you got a hold of the same board, great help. Thanks for that.

The thing is, that I have deployed 4 of these Firefly boards with the d415 attached through a USB 3.0 hub. And only one of these boards have this issue, even though I replaced both the camera and the firefly (and the ubuntu image is identical on all these 4 boards).
When my c++ software detects an issue, it reboots the OS. and afterwards call hardware_reset(). This continues many times until I detect the issue. As such it is hard for me to get a fresh dmesg-print. 

The issue occours yesterday on this one board, and now I can only see using dmesg | grep usb that the d415 seems to be registered nicely. But the "RealSense error calling rs2_pipeline_start_with_config(pipe:0x208cb5d0, config:x208b5be0): set_xu(ctrl=1) failed! Last Error: Input/output error" errors persists.

RealSense-Customer-Engineering, 2018-10-19:
[Realsense Customer Engineering Team Comment]
@BGA3 What's the kernel version on the board you used?

BGA3, 2018-10-22:
Im running version 4.4.114.

RealSense-Customer-Engineering, 2018-10-25:
[Realsense Customer Engineering Team Comment]
@BGA3 From your previous description, you only got this issue on one of 4 firefly boards, right? If so, it might be the hardware problem for this specific firefly board.  And could you please try usb2 on this issue board to see if you still get the issue?

BGA3, 2018-10-26:
Yeah actually I got the issue on 2 out of 4 boards. And yes, there seems to be an issue with the USB both in terms of power and data transfer. 

One week ago I changed all four boards to using the USB 2 post (which is okay because I dont need a high fps). So far this has been running smoothly.

I guess we can close this issue, as I guess we can blame the firefly board for messing up the camera operation. 

Thanks for the support guys.

RealSense-Customer-Engineering, 2018-10-29:
[Realsense Customer Engineering Team Comment]
@BGA3 Thanks for the update! Will close the ticket accordingly.