realsenseai/librealsense #2515: RealSense Viewer fails to start windows7

Dom3442, 2018-10-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win 7 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.16.1 }                         | 
| Language                            |             English             | 
| Segment			|                       | 

### Issue Description
I am unable to get Intel RealSense Viewer to start after installation on windows 7.  Installation seem to work ok and cameras is not showing having drivers properly installed however I'm unable to run the RealSense Viewer program to see images from the camera.  I also attempted to connect to camera using python but this also failed.

I've managed to connect to same camera on a different computer running windows 10 using the same method so assume it is related to windows 7 things.

dorodnic, 2018-10-11:
What about other examples like save-to-disk? I suspect there might be an issue with OpenGL drivers on the system, please make sure to install the latest GPU drivers for your vendor. 
Any specific error message from either the Viewer or python?

Dom3442, 2018-10-11:
Thanks for getting back to me.  I will check GPU drivers on machine.

No error message on Viewer just nothing at all happens when try to launch it.  

import pyrealsense2 as rs
pipeline = rs.pipeline()

Generates following output in pycharm
Process finished with exit code -1073740940 (0xC0000374)

Can also generate the following mesage using python from cmd line and trying to debug problem.

Unhandled exception at 0x0000000076FFF302 (ntdll.dll) in python.exe: 0xC0000374: A heap has been corrupted (parameters: 0x0000000077067C70).

Dom3442, 2018-10-12:
Hi thanks for your advice.  Updating graphics card driver did solve problem of nto being able to open RealSense Viewer

dorodnic, 2018-10-12:
Glad it works now. We need to try to reproduce and look into the python issue.

RealSense-Customer-Engineering, 2018-10-16:
[Realsense Customer Engineering Team Comment]
Hi @dom3442,

Can you create another issue to track the app fails in python on Win7 and have below info?
Is the python issue based on the prebuilt pyrealsense2 library or built from the source? which python version (2.x or 3.x) used?

Dom3442, 2018-10-16:
Hi

I've managed to solve the problem with python.  I installed the package via pip.

Uninstalling and reinstalling the package using pip worked to solve the problem and it is now working.  I'm using python 3.

Dominic

closed 2018-10-16 by Dom3442

dorodnic, 2018-10-18:
Hi @Dom3442 
Glad you managed to find a solution. The embarrassing thing is that by publishing pip package for Windows 7 unintentionally broke everyone else (#2536)... So for now we are going to roll-back Windows 7 support from pip, and I wanted to give you a heads up. 
You should still be able to use the version you downloaded, and the version in next installer should work as well, but if you encounter a problem feel free to reopen this issue or open a new one.