realsenseai/librealsense #2516: Timestamp overflow using uvc header metadata kernel patch

fubsan, 2018-10-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D430 | 
| Firmware Version                   |  5.10.03.00 | 
| Operating System & Version |   Yocto Sumo 2.5.1 with updated librealsense version 2.16.0   | 
| Kernel Version (Linux Only)    |  4.14.66-1-lts                                        | 
| Platform                                 | Intel Nuc  |
| SDK Version                            |  2.16.0                          | 
| Language                            |  C++                           | 
| Segment			|  others                           | 

### Issue Description

The public function our code calls from librealsense2 is `rs2::video_frame::gettimestamp()`
When the uint32 microsecond value in `md->header.timestamp` is converted to double, an overflow occurs and timestamps start going backwards, corrupting the video.

We fall back to the system time stamps as a work around, albeit from what I've seen, this is not recommended.

dorodnic, 2018-10-11:
Hi @fubsan 

In what way does it corrupt the video stream?
We had some discussions on this subject and decided to leave the hardware timestamp as is, and so far seen no problems with it.
We believe the primary usage of the hardware timestamps is for comparison with other hardware timestamps, and there a simple `(x - y) % max < delta` predicate is doing the work.
For single global time, we do also provide system time via frame metadata.

The main technical consideration against expanding timestamp to 64-bit is:
Let's say you start several streams, right about time of the hardware wrap-around. If first frame arrives before the wrap-around, the timestamp for that stream will get expanded, but if it arrives after, there is no way to know there was a wrap-around, so the timestamp will not get expanded.
This will create a constant gap of 1h 10min between the streams screwing with synchronization and rest of the logic. 
Obviously, this can be solved in software, but it quickly becomes frustratingly difficult requiring to keep and manage the state of all sensors (who started when)

If we could better understand your use-case, we are open to change this. Alternatively, you can see what we are doing in the timestamp_reader and propose specific modifications.

sogartar, 2019-02-01:
The best to do in this situation is to have a thread safe function where all frames go through. This function fixes their timestamps when it detects a wrap around. Here the assumption is that frames have just been captured and their timestamps should be pretty close.
This is a pretty standard approach when dealing with small clock accumulators.
The only problem would be if you don't have a frame delivered for more than the wrap around period. Then you would miss one or more wrap arounds. This use case seems very uncommon.
You can also have the timestamp in the clock's actual frequency. I bet it is not 1 MHz really. Then the wrap around would happen less often.

ev-mp, 2019-05-20:
@sogartar , in the latest v2.22.0 a new global timestamp was introduced that address the specific issue.
The PR was initially intended for inter-cam synchronization, but it also useful in the situation you describe. 
https://github.com/IntelRealSense/librealsense/pull/3909

dorodnic, 2019-06-03:
@sogartar @fubsan - thank you for suggesting this enhancement, we hope you are satisfied with the solution. If you have any follow-ups on this, feel free to open new issues.

closed 2019-06-03 by dorodnic