realsenseai/librealsense #2517: Custom firmware for D4xx

mprat, 2018-10-11:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D4xx | 
| SDK Version                            |  2                          | 
| Language                            |  C++                         | 

### Issue Description

We have a unique environment we would like to use the Realsense cameras in. Is there any documentation about the D4 processor and how the firmware interacts with it? I know the firmware is not open-sourced, but how can we find out more about to write our own custom firmware for the camera? We would love to continue using librealsense so any information is appreciated.

agrunnet, 2018-10-11:
Sorry l, but we do not open source or share FW. The only way to interact with the asic is through the open source libRealsense. Tell us what you are trying to do and we may add that capability or we may already have it on our roadmap for FW updates.

agrunnet, 2018-10-11:
Adding custom imagers is heavy lifting and FW change. I suggest you contact RealSense sales and customer enabling about your needs. As you say there may already be solutions that address your needs.

mprat, 2018-10-11:
Understood, thank you.

What about if all we want to do is integrate lenses or filters? Does the existing firmware support that use case, or is that something we need to write ourselves? Of course we will need to re-calibrate our cameras, but that is done through the official calibration tool.

agrunnet, 2018-10-11:
In principle changing lenses and baseline is not a big issue, if you build your own modules according to our specs. No FW change needed. However there are certain constraints on what lenses will work and we have different engagement models for supporting customers that range from low touch to high touch. This conversation is best had with our sales and customer enabling team in direct email or conference call.

agrunnet, 2018-10-11:
Btw you should be able to use optical filters on the outside of the lens. If these are optically flat you will not need recalibration.

MartyG-RealSense, 2018-10-11:
Coincidentally in another support case, I was just talking about micro-thin films for lenses to change their properties.

https://www.whatdigitalcamera.com/technology_guides/anti-reflection-lens-coating-work-60561

mprat, 2018-10-11:
Thank you all! I will engage with customer enabling / sales to see what we can do.

closed 2018-10-19 by RealSense-Customer-Engineering