realsenseai/librealsense #2518: librealsense 2.15 for ubuntu?

hetzbh, 2018-10-11:
Hi,

I'm looking to download/install librealsense **2.15** (not the 2.16) on Ubuntu 16.
All I found is the 2.16 version.

How/where can I download those debs (or other method to download the binaries)?

Thanks

RealSense-Customer-Engineering, 2018-10-11:
[Realsense Customer Engineering Team Comment]
Hello @hetzbh

When viewing LibRealSense github:  https://github.com/IntelRealSense/librealsense

Under "Branch: master", click "Tags", and scroll down to the LibRealSense version of your choice.  From there you will be able to download source code and binaries.

Here is version 2.15:  https://github.com/IntelRealSense/librealsense/tree/v2.15.0

I hope this helps!
Thank you!

hetzbh, 2018-10-11:
I looked, I don't see any deb files.

dorodnic, 2018-10-11:
Hi @hetzbh 
As @RealSense-Customer-Engineering suggested, the simplest and probably safer way is to fetch the tag from github and build from source.
I assume you are interested in the Debian package, however. In that case `sudo apt-get install librealsense2=2.15.0-0~realsense0.8` should work.

RealSense-Customer-Engineering, 2018-10-12:
[Realsense Customer Engineering Team Comment]
@hetzbh,

Were you able to download the .deb file you wanted with @dorodnic suggestions?

If no response we will close this issue as solved.
Thank you!

hetzbh, 2018-10-12:
I'll check it this Sunday and I'll update this ticket.

hetzbh, 2018-10-14:
`# apt-get install librealsense2=2.15.0-0~realsense0.8
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Version '2.15.0-0~realsense0.8' for 'librealsense2' was not found`

Doesn't work...

ev-mp, 2018-10-14:
Hi @hetzbh , you can use
>apt-cache showpkg librealsense2

to list all the versions available on the server, then select a tag (e.g. 2.15.0-0~realsense0.83)
>sudo apt-get install librealsense2=2.15.... 

to get the required package

hetzbh, 2018-10-14:
You're right, forgot about it (I'm a CentOS guy ;) )
One thing is missing, though, where is the librealsense2-dkms package?

ev-mp, 2018-10-14:
It is in the same location as the other librealsense packages:
> apt-cache showpkg librealsense2-dkms

Though I would recommend always have the latest version installed:
> sudo apt install librealsense2-dkms

ev-mp, 2018-10-14:
The librealsense dkms package is developed in parallel to librealsense SDK packages, but managed as a separate entity with distinct versions and update periods.

The version numbers for dkms and SDK are not overlapping.

RealSense-Customer-Engineering, 2018-10-24:
[Realsense Customer Engineering Team Comment]
Hello @hetzbh,

Were you still having issues retrieving the specific LRS 2.15 package you are looking for?

@ev-mp is correct that the dkms package is updated every SDK release and the package name doesn't change.  Command "sudo apt install librealsense2-dkms" should be sufficient for that package.

Thank you!

RealSense-Customer-Engineering, 2018-10-31:
[Realsense Customer Engineering Team Comment]
Hell again @hetzbh,

Since no response was given in past 5 days, we will go ahead and submit this issue as solved.

Thank you!

closed 2018-10-31 by RealSense-Customer-Engineering

j3cc, 2018-11-28:
@RealSense-Customer-Engineering 
Just a quick one: 
I want to chime in on this one as I am having issues retrieving legacy releases too. if I do as suggested, intel 2.16.1-0~realsense0.42 is the oldest I can retrieve. It would come in handy being able to pull version 2.15.0 as the build process is taking a lot of time and I don't know if my machine makes it through the night.

Thanks in advance