realsenseai/librealsense #2520: How to use the function colorlizer in matlab or python?

YSH6, 2018-10-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2018-10-12:
Would the information linked to below about colorizing in MATLAB be of any use to you, please?

https://communities.intel.com/message/556782#556782

YSH6, 2018-10-12:
> Would the information linked to below about colorizing in MATLAB be of any use to you, please?
> 
> https://communities.intel.com/message/556782#556782

I just want to visualize depth data in matlab ,and I need the type "White to Black",not "Jet".
Is there any easy way to do it?
thank you.

MartyG-RealSense, 2018-10-12:
You can select White To Black mode from a drop-down menu in the RealSense Viewer or set it with scripting.

https://communities.intel.com/message/562179#562179

YSH6, 2018-10-12:
> You can select White To Black mode from a drop-down menu in the RealSense Viewer or set it with scripting.
> 
> https://communities.intel.com/message/562179#562179

I selected White To Black mode and recorded.Then I use "realsense.rosbag_example" to read the .bag file,
which shows in mode Jet like that
[image]
Still in mode Jet
I know there is a class colorizer and it can be used well in C++
[image]
How can I use it in matlab?

MartyG-RealSense, 2018-10-12:
Apologies for the delay in responding, I was investigating your case further.

In my own test, although the bag defaulted back to Jet when opened in the RealSense Viewer, I was able to change the bag back to White To Black by selecting this mode again in the Depth Visualization section of the Viewer's options panel.

You can playback a bag in the Viewer instead of rosbag_example by going to the Add Source menu at the top of the Viewer's options panel and selecting the Load Recorded Sequence option.  Or you can just drag-and-drop the bag file onto the Viewer's center panel.

MartyG-RealSense, 2018-10-12:
It is possible you may also be able to change the visualization mode through the SDK's MATLAB wrapper, but I do not have the scripting code for doing that.  Hopefully one of the Intel team can help you with that later.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/matlab

dorodnic, 2018-10-12:
Hi @YSH6 @MartyG-RealSense 
Based on user report (#2487) there seems to be a bug in setting processing options right now for the Matlab wrapper. We are looking into it.
This can be done with the python wrapper, by calling `colorizer.set_option` and choosing a different color-scheme. 
That said, both python (via numpy and opencv) and Matlab have plenty of work-around ways to do this.

dorodnic, 2018-10-12:
Certainly:
```python
import matplotlib.pyplot as plt   
import pyrealsense2 as rs   
...
colorizer = rs.colorizer()
colorizer.set_option(rs.option.color_scheme, 2); // white to black
colorized_depth = np.asanyarray(colorizer.colorize(depth_frame).get_data())
plt.imshow(colorized_depth)
```
There is no closed enumeration for the different color schemes, but you can play around with different values and other options. You can also test it in an interactive notebook [image]

dorodnic, 2018-10-12:
We also have [additional notebook](https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_filters.ipynb) explaining post-processing and post-processing options, but it is missing the colorizer options

YSH6, 2018-10-12:
> Certainly:
>
> ```python
> import matplotlib.pyplot as plt
> import pyrealsense2 as rs
> ...
> colorizer = rs.colorizer()
> colorizer.set_option(rs.option.color_scheme, 2); // white to black [...]
That works,
I really appreciate your help

RealSense-Customer-Engineering, 2018-10-19:
[Realsense Customer Engineering Team Comment]
@ysh6 Any other questions about this ticket? Please update. Thanks!

lramati, 2018-10-29:
PR #2492 Added the missing constructor argument to matlab's colorizer class, so the ability to set the color scheme should be working there too

RealSense-Customer-Engineering, 2018-10-30:
[Realsense Customer Engineering Team Comment]
@YSH6 Any other questions for this ticket? Please update. Thanks!

YSH6, 2018-10-30:
> [Realsense Customer Engineering Team Comment]
> @YSH6 Any other questions for this ticket? Please update. Thanks!
Apologies for the delay in responding,My problem was settled.

YSH6, 2018-10-30:
> PR #2492 Added the missing constructor argument to matlab's colorizer class, so the ability to set the color scheme should be working there too

That's good news.
Thank you!

closed 2018-10-30 by RealSense-Customer-Engineering

Ted-EE, 2020-06-01:
> Certainly:
>
> ```python
> import matplotlib.pyplot as plt
> import pyrealsense2 as rs
> ...
> colorizer = rs.colorizer()
> colorizer.set_option(rs.option.color_scheme, 2); // white to black [...]

Sir, how can I write it in C++. I just referred from https://github.com/IntelRealSense/librealsense/issues/3283, in my code, 
`   

    rs2::colorizer color_map;

    rs2::pipeline pipe;

    rs2::config pipe_config;
    pipe_config.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
    pipe_config.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);

    rs2::pipeline_profile profile = pipe.start(pipe_config);
    const char* depth_win = "depth_Image";
    namedWindow(depth_win, WINDOW_AUTOSIZE);
    //const char* color_win = "color_Image";
    //namedWindow(color_win, WINDOW_AUTOSIZE);

    //    float depth_scale = get_depth_scale(profile.get_device());
    //    rs2_stream align_to = find_stream_to_align(profile.get_streams());

    while (waitKey(1) && cvGetWindowHandle(depth_win)) {
        rs2::frameset data = pipe.wait_for_frames();

        rs2::frame depth = data.get_depth_frame().apply_filter(color_map);
        //rs2::frame color = data.get_color_frame();

        rs2::colorizer c;
        c.set_option(RS2_OPTION_COLOR_SCHEME, 2); // white to black
        depth = c.colorize(depth);
       
        const int depth_w = depth.as<rs2::video_frame>().get_width();
        const int depth_h = depth.as<rs2::video_frame>().get_height();
        //const int color_w = color.as<rs2::video_frame>().get_width();
        //const int color_h = color.as<rs2::video_frame>().get_height();

        Mat depth_image(Size(depth_w, depth_h), CV_8UC3, (void*)depth.get_data(), Mat::AUTO_STEP);
        /*Mat color_image(Size(color_w, color_h), CV_8UC3, (void*)color.get_data(), Mat::AUTO_STEP);*/

        imshow(depth_win, depth_image);
        /*imshow(color_win, color_image);*/
`

I was trying to display only the depth image in white-to-black frame and print the matrix of depth image.

Thanks.