realsenseai/librealsense #2521: Python wrapper issue: pyrealsense2.pyrealsense2.frame object returned by decimation.process() does not expose get_distance()

sachinsdate, 2018-10-12:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.10.03.00 | 
| Operating System & Version |   Win (8.1/10)  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC |
| SDK Version                            |  SDK 2.0                          | 
| Language                            |  python                          | 
| Segment			|  Robot/Smartphone/VR/AR/others                          | 

### Issue Description

I am getting the following error when I call get_distance() on the frame object that is returned by rs.decimation_filter().process():

`
AttributeError: 'pyrealsense2.pyrealsense2.frame' object has no attribute 'get_distance'
`

**What am I doing wrong? and what are my options for fixing the issue?** 

Sorry if I using the API incorrectly. Just got my D415, Hence a newbie to RealSense :=)

Here is my code:
```python
pipeline = rs.pipeline()
config = rs.config()
rs.config.enable_device_from_file(config, bag_file_path)
config.enable_stream(rs.stream.depth)
config.enable_stream(rs.stream.color)
profile = pipeline.start(config)
depth_sensor = profile.get_device().first_depth_sensor()
depth_scale = depth_sensor.get_depth_scale()

align_to = rs.stream.color
align = rs.align(align_to)

#setup the filters
depth_to_disparity = rs.disparity_transform(True)
disparity_to_depth = rs.disparity_transform(False)
decimation = rs.decimation_filter()
spatial = rs.spatial_filter()
temporal = rs.temporal_filter()
hole_filling = rs.hole_filling_filter()

try:
	while True:
		print 'Fetching aligned frames from stream...'
		frames = pipeline.wait_for_frames()
		aligned_frames = align.process(frames)

		aligned_depth_frame = aligned_frames.get_depth_frame()
		color_frame = aligned_frames.get_color_frame()

		if not aligned_depth_frame or not color_frame:
			continue

		print 'frame: ',frame_num

		#apply the filters
		depth_frame = decimation.process(aligned_depth_frame)
		depth_frame = depth_to_disparity.process(depth_frame)
		depth_frame = spatial.process(depth_frame)
		depth_frame = temporal.process(depth_frame)
		depth_frame = disparity_to_depth.process(depth_frame)
		depth_frame = hole_filling.process(depth_frame)

		depth_image = np.asanyarray(depth_frame.get_data())
		color_image = np.asanyarray(color_frame.get_data())
		color_image = cv2.resize(color_image,dsize=(0,0), fx=0.5, fy=0.5, interpolation=cv2.INTER_AREA)

		num_rows = depth_image.shape[0]
		num_cols = depth_image.shape[1]

		print 'Re sampled frame size (r,c)=', (num_rows, num_cols)
		depth_intrin = depth_frame.profile.as_video_stream_profile().intrinsics

		for r in range(0, num_rows, 5):
			for c in range(0, num_cols, 5):
				#THE FOLLOWING CALL FAILS
				depth = depth_frame.get_distance(c, r)
				#THE FOLLOWING CALL EXECUTES FINE IF YOU PLUGIN A BOGUS DEPH
				depth_point_mts_cam_coords = rs.rs2_deproject_pixel_to_point(depth_intrin, [c, r], depth)

```

lramati, 2018-10-12:
The various post processing functions return a generic frame, so you need
to cast to depth_frame to reexpose the get_distance() function. You can do
this with `depth_frame = frame.as_depth_frame()`. Note that this will only
work on a depth frame and will fail on for example a color frame.

sachinsdate, 2018-10-12:
Thanks @lramati.  Casting down did the trick.

The following also seems to work:
```python
	depth_frame.__class__ = rs.pyrealsense2.depth_frame
```

closed 2018-10-12 by sachinsdate