realsenseai/librealsense #2522: On the accuracy of factory calibration

SirDifferential, 2018-10-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 & D435} | 
| Firmware Version                   | 5.10.3 | 
| Operating System & Version |   Win 8.1 | 
| Platform                                 | x86_64 Desktop |
| SDK Version                            | 2.16.1, official release | 
| Segment			|  Calibration  | 

### Issue Description

We've been testing several D415 and D435 cameras using the depth quality checker tool with gold calibration and self-made dynamic calibration, using the official tools and firmware 5.10.3. We tested at about 1.8 and 2.8 meters of distance in a dark-ish room with the emitter on using the default preset. The wall is white painted wood of some sort. Room's about 23°C.

We are seeing some varying levels of accuracy, mostly in the "Z Accuracy" field of the quality checker tool. Based on some other questions on these issue pages, the calibration shouldn't be necessary in most cases.

Camera 1: D435, might have been with the room lit by lights

Gold: -3.88% [pic](https://plantmonster.net/delicode/rs_calib/rs1_435/calib_gold.png)
Ours: 0.65% [pic](https://plantmonster.net/delicode/rs_calib/rs1_435/calib_new.png)

Camera 2: D435

Gold: 8.90% [pic](https://plantmonster.net/delicode/rs_calib/rs2_435/goldcalib.png)
Ours: 1.93% [pic](https://plantmonster.net/delicode/rs_calib/rs2_435/newcalib.png)

Camera 3: D415

Gold: 1.73% [pic](https://plantmonster.net/delicode/rs_calib/rs3_415/calib_gold.png)
Ours: 0.38% [pic](https://plantmonster.net/delicode/rs_calib/rs3_415/calib_new.png)

Camera 4: D415

Gold: 0.41% [pic](https://plantmonster.net/delicode/rs_calib/rs4_415/gold.png)
Ours: 2.24% [pic](https://plantmonster.net/delicode/rs_calib/rs4_415/calib_new.png)

1) Is there something we didn't account for in the benchmark? Should we take something else into account (room brightness, emitter, some other values in the tool, preset?)

2) Most of these devices have been sitting on our shelf without much tampering, so any impact based miscalibration is unlikely. What explains the large errors with some sensors? Is the sensor prone to temperature differences like Orbbec Astra (they heat up over 30 mins), or is the factory calibration itself at fault?

This is note a huge issue. All this means is that we'll have to verify each sensor individually for their error and see if it needs a manual calibration.

MartyG-RealSense, 2018-10-12:
Intel sell a professional calibration tool in their online Click store that is aimed at engineering departments and OEM factories.  It supports multiple cameras and remote access, and although it costs $1500, this reflects the cost of manufacturing it.

https://click.intel.com/realsense-d400-camera-oem-calibration.html

Intel's illustrated camera tuning guide is also an excellent document for working out where improvements can be made to imaging results.

https://realsense.intel.com/wp-content/uploads/sites/63/BKM-For-Tuning-D435-and-D415-Cameras-Webinar_Rev3.pdf

SirDifferential, 2018-10-15:
Thank you for the links. The tuning one in particular was useful.

closed 2018-10-15 by SirDifferential