realsenseai/librealsense #2524: d435 generating "gridded" depth data with python script

diu7, 2018-10-12:
dear real sense support,
we are encountering the same problem which have been posed in issue 2202. There are many grid generated in the depth image.
[https://github.com/IntelRealSense/librealsense/issues/2202](url)
it seems like that the problem have been solved with  version 2.15.
However ,when we follow the instruction and install the latest version sdk, it seems like that the problem is still existing with the python code.
[image]
[image]
[image]

dorodnic, 2018-10-12:
Hi @diu7 
I think the issue here is `pip install` hasn't been updated. We will address it in upcoming days.
In the meantime, if you build the python wrapper from source it should solve the issue for now.

diu7, 2018-10-13:
@dorodnic  Thanks！！I use the second method which you said and it was work

dorodnic, 2018-10-14:
PiPy repo updated: [pypi.org/project/pyrealsense2](https://pypi.org/project/pyrealsense2/2.16.1.272/)

closed 2018-10-19 by RealSense-Customer-Engineering