realsenseai/librealsense #2525: How to connect Multi-Camera in a computers

NguyenThaiHoc1, 2018-10-13:
Dear Realsense support,
I have some questions for you and i hope you can help to solve my problem
First: I have 3 camera (Intel-D435) and i want to connect those camera in a computer. I try to connect 2 camera in USB gate in my computer (my computer only 2 USB gate). after that, i turn on tool calling is Intel-RealSence-Viewer. But It just connected a camera and displayed a camera, i want you to help me 
(How can i connect 3 camera in a computer)

Second: the problem relate to technique. I read paper in website. RGB image is 3 channels and I think RGB-D image is 4 channels. But i follow the code providing in librealsense github (https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb)
I saw 
(RGB-Image)
color = np.asanyarray(color_frame.get_data()) 
=> Shape (640, 480, 3) it have shape = 3 
(Depth-Image)
colorizer = rs.colorizer()
colorized_depth = np.asanyarray(colorizer.colorize(depth_frame).get_data())
plt.imshow(colorized_depth)
=> Shape(1280, 640, 3) it is still shape =3 (Why ??????)
when combine RGB and Depth Image 
images = np.hstack((color, colorized_depth))
plt.imshow(images)
=> images variable have shape(1260, 640, 3) , shape is still = 3 (I don't Understand because i think shape is (1260, 640, 4)   it is 4 channels ??? )
I newbie in the field of image processing 3D (I wish i understand it  and I hope every help me   )

Thanks you,

MartyG-RealSense, 2018-10-13:
In regard to your first question: you can add cameras in RealSense Viewer by left-clicking on the Add Source option at the top of the Viewer's options panel and selecting a camera from the list that drops down.

[image]

NguyenThaiHoc1, 2018-10-13:
> In regard to your first question: you can add cameras in RealSense Viewer by left-clicking on the Add Source option at the top of the Viewer's options panel and selecting a camera from the list that drops down.
> 
> [image]

Hi @MartyG-RealSense 
I did. But i want to display others camera in the same screen computer (How to display a lot of camera  in the same app called - Intel-RealSence-Viewer ?)

Please help me ! 
sorry for my English skill is very bad

MartyG-RealSense, 2018-10-14:
Your English is very good, don't worry.

Once you have added cameras with Add Source, you should be able to open multiple stream windows by going to each camera in the side panel and left-clicking on its red icon to turn the icon blue and start the stream.

In the image below, I have opened a D415 and an SR300 camera in the Viewer and started the RGB, IR and depth streams on both cameras so that, I have 6 streams displayed.

[image]

NguyenThaiHoc1, 2018-10-15:
I did it 
Thanks you a lot @MartyG-RealSense

closed 2018-10-15 by NguyenThaiHoc1

sdemikos, 2019-03-26:
Dear Realsense support,
I need a camera to read the movement of people in the space for installation. I am thinking of getting a Realsense camera however before that I would like to know if 
 I can connect a Realsense camera to a MAC 2011 El Capitan. or I need a PC?
 2 question: Can I connect a Realsense camera to MAX /MSP?
Please let me know
 Thanks 
 Silvana