realsenseai/librealsense #2526: PyLibRealsense Saving Aligned PointCloud

tmxosvhem, 2018-10-13:
Hi,

I see there is an example for creating a pointcloud from color and depth frames and saving it as a .ply (export_ply_example.py), and I also see an alignment example (align_depth2color.py).

But, is there an example for saving an aligned pointcloud as a .ply file?

I took a piece of code from export_ply_example.py (marked in red) and put it into align_depth2.color.py as shown below.

```
pc = rs.pointcloud()
points = rs.points()

# Streaming loop
try:
    while True:
        # Get frameset of color and depth
        frames = pipeline.wait_for_frames()
        # frames.get_depth_frame() is a 640x360 depth image
        # Align the depth frame to color frame
        aligned_frames = align.process(frames)

        # Get aligned frames
        aligned_depth_frame = aligned_frames.get_depth_frame() # aligned_depth_frame is a 640x480 depth image
        color_frame = aligned_frames.get_color_frame()

        pc.map_to(color_frame)
        # Generate the pointcloud and texture mappings
        points = pc.calculate(aligned_depth_frame)
        print("Saving to 1.ply...")
        points.export_to_ply("1.ply", color_frame)
        print("Done")

        # Validate that both frames are valid
        if not aligned_depth_frame or not color_frame:
            continue

        depth_image = np.asanyarray(aligned_depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
```
 
However, the resulting pointcloud is still not aligned.

Best Regards,

dorodnic, 2018-10-14:
Hi @tmxosvhem 
What do you mean by point-cloud being aligned / not aligned?

tmxosvhem, 2018-10-14:
If I just run export_ply_example.py to create a .ply file and open it in Meshlab, the color and depth seemed to be misaligned. On the other hand if I save a .ply from the Realsense Viewer I do not see the issue and looks perfectly fine.

dorodnic, 2018-10-14:
Are you using `pyrealsense` from `pip install`? If so, could you try running `pip install pyrealsense2` again? We have updated the repo with a newer version today

RealSense-Customer-Engineering, 2018-10-19:
[Realsense Customer Engineering Team Comment]
@tmxosvhem Did you get chance to try as Dorodnic suggested? Looking forward to your update. Thanks!

closed 2018-10-23 by tmxosvhem