realsenseai/librealsense #2529: Error trying to run box_dimensioner_multicam

donmerchant, 2018-10-15:
I'm using the following:
Intel NUC7i3BNH 8GB ram, 128GB SSD
Ubuntu 18.04.1 LTS 
Kernel 4.15.0-36-generic
Python 3.6.6 
Here's what I get when I run this:
`snuc@snuc-NUC:~/box$ python3 box_dimensioner_multicam_demo.py
1 devices have been found
Traceback (most recent call last):
  File "box_dimensioner_multicam_demo.py", line 141, in <module>
    run_demo()
  File "box_dimensioner_multicam_demo.py", line 75, in run_demo
    if not transformation_result_kabsch[device][0]:
KeyError: '821212060318'`

What causes this and how do I fix it? Thanks.

RealSense-Customer-Engineering, 2018-10-22:
[Realsense Customer Engineering Team Comment]
Hi @donmerchant,

Please check the 1st point of workflow in the code. One chessboard is required and its checkboard parameters (width, height, size) need to be updated accordingly.

RealSense-Customer-Engineering, 2018-11-28:
[Realsense Customer Engineering Team Comment]
Hi @donmerchant,

Still need to support for this ticket? If not, will close this.

closed 2018-12-17 by RealSense-Customer-Engineering