realsenseai/librealsense #2535: Viewer dropping frames during recording

aman-tiwari, 2018-10-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | 5.10.06.00 | 
| Operating System & Version |  Win 10  | 
| Kernel Version (Linux Only)    |  N/A                                    | 
| Platform                                 | PC |

+SDK Version                            |  2                           | 
| Language                            |  N/A                          | 
| Segment			|  others                           | 

### Issue Description
It seems like the viewer is dropping frames when trying to record 720p depth and 720p rgb streams at the same time. The FPS seems to stay ~30fps but the resultant .bag files are choppy when played back. Are there any settings to tweak (e.g the pipeline queue size) to improve the performance? 
Note that I also observe frame drops when recording to a file using `pipeline.enable_record_to_file`.

dorodnic, 2018-10-16:
Hi @aman-tiwari 
If you build from source, please ensure you are using Release version.
If you are using pre-compiled SDK, the most likely cause is slow hard-drive.

aman-tiwari, 2018-10-16:
I'm using the latest precompiled binaries, but I'm writing to an NVMe SSD, which should be more than fast enough

aman-tiwari, 2018-10-16:
Also, I don't get frame drops if I do the saving to disk myself (e.g by using the SDK and passing the frames to a different thread pool to save them). But then I lose the ability to change postprocessing and visualisation optoions of the data that the ros bag gives.

RealSense-Customer-Engineering, 2018-10-19:
[Realsense Customer Engineering Team Comment]
@aman-tiwari
Seems to be similar issue as #2436 https://github.com/IntelRealSense/librealsense/issues/2436 . Could you please try below two options to see if you still get the issue? Thanks!

1. Disable "Auto-Exposure" and set it manually, verifying it does not exceed the upper bound.
2. Leave "Auto-Exposure switched on, but disable "Auto-Exposure Priority" control to enforce constant FPS.

aman-tiwari, 2018-10-23:
Changing the priority seemed to help reduce the number of dropped frames. Manually setting the exposure worked. In the end I found that it recorded much more smoothly on Ubuntu and switched to an installation of that.

closed 2018-10-29 by RealSense-Customer-Engineering