realsenseai/librealsense #2536: Device not detected by pyrealsense2==2.16.1.272

aman-tiwari, 2018-10-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.10.06.00 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  n/a                                  | 
| Platform                                 |  |
| SDK Version                            |  2.16.1.272                         | 
| Language                            |   python                       | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

The latest version of `pyrealsense2` doesn't detect my D435, but the previous version did. The viewer also can detect the device.

RealSense-Customer-Engineering, 2018-10-16:
[Realsense Customer Engineering Team Comment]
Hi @aman-Tiwari,

What's your previous working version? Did you use the pre-built pyrealsense2 library or build it from the source?

aman-tiwari, 2018-10-16:
The previous working version I used was 2.14.1.91, from a pip install.

dorodnic, 2018-10-18:
We found the problem and working to fix it right now.

dorodnic, 2018-10-18:
The problem should now be resolved, please run:
```
pip uninstall pyrealsense2
pip install pyrealsense2
```
The root cause was Windows7 version overwriting the regular package.

closed 2018-10-24 by dorodnic