realsenseai/librealsense #2538: How to measure the diameter of a circle?//怎么测量圆直径？

laolaolulu, 2018-10-16:
In the RGB stream, using OpenCV's HoughCircles to find a circle, I want to measure the diameter of the circle.
//RGB流中使用opencv的HoughCircles 找到一个圆，我想测量这个圆的直径多少

RealSense-Customer-Engineering, 2018-10-17:
[Realsense Customer Engineering Team Comment]
Hi @laolaolulu,

See #2382 as a reference. You can align first, and then measure the distance between the center and the point on the circle you find.

RealSense-Customer-Engineering, 2018-10-17:
[Realsense Customer Engineering Team Comment]
You can also refer to rs-measure app to implement.
https://github.com/IntelRealSense/librealsense/tree/master/examples/measure

laolaolulu, 2018-10-19:
I can't calculate the actual size using align = rs. align (rs. stream. color); using align = rs. align (rs. stream. depth) can calculate the size, but getting_color_frame () can't use opencv operation in the image;
//我使用align = rs.align( rs.stream.color)无法计算出实际尺寸；使用align = rs.align(rs.stream.depth)能计算尺寸但是get_color_frame()获取的图片中无法使用opencv操作；
        vdist1 = aligned_depth_frame.get_distance(300,450)
        vdist2 = aligned_depth_frame.get_distance(500,450)
        cv2.circle(img, (300,450), 2, (0, 100, 100), 3)
        cv2.circle(img, (500,450), 2, (0, 100, 100), 3)
        intr = aligned_depth_frame.profile.as_video_stream_profile().intrinsics
        dep1 = rs.rs2_deproject_pixel_to_point(intr,[300,450], vdist1)
        dep2 = rs.rs2_deproject_pixel_to_point(intr,[500,450], vdist2)
        c = np.abs(dep1[0] - dep2[0])
        print(c)

RealSense-Customer-Engineering, 2018-10-19:
[Realsense Customer Engineering Team Comment]
Hi @laolaolulu,

What do you mean you can't operate the color data from "get_color_frame()"? Did you correctly convert to cv::Mat?

You can use opencv to find the circle pixels in the color image before color is aligned to depth. Then, use the deproject pixel to point function to find the 3D point.

RealSense-Customer-Engineering, 2018-10-30:
[Realsense Customer Engineering Team Comment]
HI @laolaolulu,

any update?

RealSense-Customer-Engineering, 2018-11-14:
[Realsense Customer Engineering Team Comment]
Hi @laolaolulu,

Will close this if no further question.

closed 2018-11-14 by laolaolulu