realsenseai/librealsense #2540: Using playback function in unity crash

rouk1, 2018-10-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |                           
| Platform                                 | Windows 10 |
| SDK Version                            |  v2.16.1 }                          | 
| Language                            |  unity }                          | 

### Issue Description
- download latest unitypackage (v2.16.1)
- record a .bag file using RealSense Viewer 
- play it in any exemple of the unitypackage

Here are the editor logs.
[Editor.log](https://github.com/IntelRealSense/librealsense/files/2482803/Editor.log)
[error.log](https://github.com/IntelRealSense/librealsense/files/2482804/error.log)
[crash.dump](https://dl.dv.fr/25FO)

ogoshen, 2018-11-11:
Might be related to #2472

When using the processing pipe Unity can crash with:
```
ArgumentException: GCHandle value belongs to a different domain
  at System.Runtime.InteropServices.GCHandle.op_Explicit (IntPtr value) [0x0002b] in /Users/builduser/buildslave/mono/build/mcs/class/corlib/System.Runtime.InteropServices/GCHandle.cs:127 
  at System.Runtime.InteropServices.GCHandle.FromIntPtr (IntPtr value) [0x00000] in /Users/builduser/buildslave/mono/build/mcs/class/corlib/System.Runtime.InteropServices/GCHandle.cs:172 
  at Intel.RealSense.CustomProcessingBlock.ProcessingBlockFrameCallback (IntPtr f, IntPtr u) [0x00000] in <filename unknown>:0 
  at (wrapper native-to-managed) Intel.RealSense.CustomProcessingBlock:ProcessingBlockFrameCallback (intptr,intptr)
  at (wrapper managed-to-native) Intel.RealSense.NativeMethods:rs2_synthetic_frame_ready (intptr,intptr,object&)
  at Intel.RealSense.FrameSource.FrameReady (IntPtr ptr) [0x00000] in <filename unknown>:0 
  at Intel.RealSense.FrameSource.FrameReady (Intel.RealSense.Frame f) [0x00000] in <filename unknown>:0 
  at RsProcessingPipe.ProcessFrame (Intel.RealSense.Frame frame, FrameSource src) [0x00081] in D:\librealsense\wrappers\unity\Assets\RealSenseSDK2.0\Scripts\RsProcessingPipe.cs:103 
UnityEngine.DebugLogHandler:Internal_LogException(Exception, Object)
UnityEngine.DebugLogHandler:LogException(Exception, Object)
UnityEngine.Logger:LogException(Exception, Object)
UnityEngine.Debug:LogException(Exception)
RsProcessingPipe:ProcessFrame(Frame, FrameSource) (at Assets\RealSenseSDK2.0\Scripts\RsProcessingPipe.cs:110)
Intel.RealSense.CustomProcessingBlock:ProcessingBlockCallback(IntPtr, IntPtr, IntPtr)
Intel.RealSense.NativeMethods:rs2_process_frame(IntPtr, IntPtr, Object&)
Intel.RealSense.CustomProcessingBlock:ProcessFrame(Frame)
RsDevice:RaiseSampleEvent(Frame) (at Assets\RealSenseSDK2.0\Scripts\RsDevice.cs:146)
RsDevice:WaitForFrames() (at Assets\RealSenseSDK2.0\Scripts\RsDevice.cs:159)
```
Which actually means that `GCHandle.FromIntPtr` got an invalid pointer, probably because the handle was already freed.

Need to make sure that the `CustomProcessingBlock` instance stays alive to complete the callback.

matkatz, 2018-12-04:
Sorry for the long delay.
The Playback issue was handled by PRs #2646, #2598 which were merged to the master branch and are available on release v2.17.0e should