realsenseai/librealsense #2542: Question About align depth to color

laihaotao, 2018-10-16:
Hi, recently I am doing a depth and color alignment, and would like to know exactly it works. It found the `align()` method inside the `align.cpp` file. And notice the following code 

```c++
// ############ No.1
float depth_pixel[2] = { depth_x - 0.5f, depth_y - 0.5f }, depth_point[3], other_point[3], other_pixel[2];
rs2_deproject_pixel_to_point(depth_point, &depth_intrin, depth_pixel, depth);
rs2_transform_point_to_point(other_point, &depth_to_other, depth_point);
rs2_project_point_to_pixel(other_pixel, &other_intrin, other_point);
// ############ No.2
const int other_x0 = static_cast<int>(other_pixel[0] + 0.5f);
const int other_y0 = static_cast<int>(other_pixel[1] + 0.5f);
```

But I don't really understand the code above why it need to first do `depth_x - 0.5f` and `other_pixel[0]+0.5` in the two marker positions.

Any help is highly appreciated. Thanks!

dorodnic, 2018-10-16:
@aangerma please comment

aangerma, 2018-10-17:
Hello  @laihaotao 
If the resolution of the other image is bigger then the depth resolution, when we will map from depth to the other image not all the pixels of the other will have corresponding pixel from depth, it means that the aligned depth image will have 'holes' so, we are taking two of the corners of each pixel in depth (top-left and bottom-right), map them to the other image and fill all the area with the value of this pixel so we will cover all the pixels of the other image. 
hope its more clear now.

laihaotao, 2018-10-17:
> Hello @laihaotao
> If the resolution of the other image is bigger then the depth resolution, when we will map from depth to the other image not all the pixels of the other will have corresponding [...]

Thank you so much for you reply, it is extremely helpful.

dorodnic, 2018-10-17:
I'd also add a comment about how it relates to lines [align.cpp:319-325](https://github.com/IntelRealSense/librealsense/blob/master/src/proc/align.cpp#L319-L325) and the strange gridded artifact people have been reporting around version 2.15:
If we **could** know that the mapping is 1-to-1, this would **help greatly** since we could calculate two times less projection operations. Sadly the heuristic is not straightforward, it is a complex formula of the pixel density as function of field-of-view and resolution. Getting this right can speed things up significantly, but getting it wrong will result in black lines where depth is supposed to be "stretched".

laihaotao, 2018-10-25:
thanks for your reply. solved

closed 2018-10-25 by laihaotao