realsenseai/librealsense #2543: No Frames Received when using multiple cameras

fwindolf, 2018-10-16:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 (D435, D415) | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |  Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.15.0-36.39~16.04.1-generic 4.15.18 | 
| Platform                                 | PC  |
| SDK Version                            |  2.16.1                        | 
| Language                            |  C                        | 

### Issue Description
When starting streams from more than one camera in the realsense-viewer, the last started device will fail to produce frames in all resolutions (both with custom and prebuilt realsense-viewers).
However, the rs-multicam application works just fine. 

With realsense-viewer:
[image]

With rs-multicam:
[image]

dorodnic, 2018-10-16:
On the 3D tab we have a "lock" button, enabled by default. Please try disabling it and see if it helps. 
There is an existing bug we are about to roll the fix for in couple of days that sounds similar to this. 
If this is not a software bug, I'd double check the resolutions in multicam example, but I strongly suspect it is.

fwindolf, 2018-10-16:
Disabling the lock button in the 3D pane helped, now I am able to see color streams in highest resolution without frame drops. 
If I additionally add the depth streams for both cameras, there is loads of incomplete frames. I assume that is standard behaviour as the bandwidth is too low to support 1920*1080 color and 1280*720 depth in parallel?

dorodnic, 2018-10-16:
> Disabling the lock button in the 3D pane helped

Good, the fix will be released shortly. We apologize for the issue.

> If I additionally add the depth streams for both cameras, there is loads of incomplete frames

In the multicam example we are not displaying the incomplete frames warnings, but they are probably still there. Most likely related to USB bandwidth, but it is also a good idea to make sure you are using Release version (or the pre-compiled version from `apt-get`)

fwindolf, 2018-10-17:
> Good, the fix will be released shortly. We apologize for the issue.
No worries, thanks for the quick help!
>  Most likely related to USB bandwidth, but it is also a good idea to make sure you are using Release version
I used the release version. The issue seems to be on my side, as I checked on my home PC with 3 (D415, D415, D435) cameras and there are no incomplete frames (only shortly after activating the streams) on highest resolution for both depth and color, even though the cameras are on the same bus.
```
/sys/devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2:1.0/video4linux/video0
/sys/devices/pci0000:00/0000:00:14.0/usb4/4-1/4-1:1.0/video4linux/video3
/sys/devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3:1.0/video4linux/video6
```

closed 2018-10-17 by fwindolf