realsenseai/librealsense #2547: Undefined symbol with Python wrapper

FlavGal, 2018-10-17:
Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   |05.10.03.00 | 
| Operating System & Version |  Linux (Ubuntu Mate 16.04)  | 
| Kernel Version (Linux Only)    |  4.14.17-v7+                                         | 
| Platform                                 | Raspberry Pi  |
| SDK Version                            |  2.16.1                          | 
| Language                            |  Python                         | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hello, I am trying to use the Python wrapper that I compiled following the guide but it doesn't seem to work. I tried to use the .so files directly in the same folder as the exemple script given with the installation procedure or in the PYTHONPATH but everytime I have the same error : 

``  File "test.py", line 1, in <module>
    import pyrealsense2 as rs
ImportError: /usr/local/lib/pyrealsense2.so: undefined symbol: PyInstanceMethod_Type ``

lramati, 2018-10-17:
Are you running the same version of python as you compiled for? The cmake
script defaults to python 3 if it's installed, while some Linux distros
still default to python 2 on the command line

>

mksarker, 2019-09-19:
I have the same issue. I am new here. What you did to solve this issue @FlavGal, Please give me some details. Thanks in advance!

lramati, 2019-09-19:
Are you sure you compiled the module for the correct version of Python? When you have both Python 2.X and 3.X installed, CMake defaults to building for Python 3

mksarker, 2019-09-19:
Now working! Thanks.

surefyyq, 2020-11-15:
@lramati there are python2 and python3 in my ubuntu, but it seems to make using python2, how can I control to make using python3?
Thanks