realsenseai/librealsense #2552: How to get a aligned depth image and RGB image with the resolution of 1280*720

fu3lab, 2018-10-17:
Hi
We can align the RGB image with the depth image, but they are only can be 640*480. How can I get a aligned depth image and RGB image with the resolution of 1280*720? Is there any parematers in the aligh class?
Thanks.
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (05.09.14.00) | 
| Operating System & Version |   {Win 10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<16>.<1> }                          | 
| Language                            |  {C++ }                          | 
| Segment			| {others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

fu3lab, 2018-10-17:
The images are attached.
[image]
[image]

MartyG-RealSense, 2018-10-17:
A simple Align script can be found here:

https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#frame-alignment

You may also find the rs-align sample program useful, as it generates a 1280x720 window for alignment to take place in.

https://github.com/IntelRealSense/librealsense/tree/master/examples/align

fu3lab, 2018-10-17:
Thanks. I can display the aligned images in 1280 * 720, but cannot save them as displayed. I would like to know how to save the aligned images in 1280 * 720?
[image]

MartyG-RealSense, 2018-10-17:
One of the Intel guys should be able to help with this.  Thanks for your patience!

MartyG-RealSense, 2018-10-18:
The Intel support members on this GitHub do their best to answer the majority of questions in a timely manner, though some cases take longer than others.  I promise that as long as your case remains open, Intel are aware of it.

fu3lab, 2018-10-18:
#include <librealsense2/rs.hpp>
#include "../example.hpp"
// 3rd party header for writing png files
#define STB_IMAGE_WRITE_IMPLEMENTATION
#include "stb_image_write.h"

#include <sstream>
#include <iostream>
#include <fstream>
#include <algorithm>
#include <cstring>

void remove_background(rs2::video_frame& other, const rs2::depth_frame& depth_frame, float depth_scale, float clipping_dist);
float get_depth_scale(rs2::device dev);
rs2_stream find_stream_to_align(const std::vector<rs2::stream_profile>& streams);
bool profile_changed(const std::vector<rs2::stream_profile>& current, const std::vector<rs2::stream_profile>& prev);

int main(int argc, char * argv[]) try
{
	// Create and initialize GUI related objects
	window app(1280, 720, "CPP - Align Example"); // Simple window handling     // ImGui library intializition
	rs2::colorizer c;                          // Helper to colorize depth images
	texture renderer;                     // Helper for renderig images

										  // Create a pipeline to easily configure and start the camera
	rs2::pipeline pipe;
	//Calling pipeline's start() without any additional parameters will start the first device
	// with its default streams.
	//The start function returns the pipeline profile which the pipeline used to start the device
	rs2::pipeline_profile profile = pipe.start();

	// Each depth camera might have different units for depth pixels, so we get it here
	// Using the pipeline's profile, we can retrieve the device that the pipeline uses
	float depth_scale = get_depth_scale(profile.get_device());

	//Pipeline could choose a device that does not have a color stream
	//If there is no color stream, choose to align depth to another stream
	rs2_stream align_to = find_stream_to_align(profile.get_streams());

	// Create a rs2::align object.
	// rs2::align allows us to perform alignment of depth frames to others frames
	//The "align_to" is the stream type to which we plan to align depth frames.
	rs2::align align(align_to);

	// Define a variable for controlling the distance to clip
	float depth_clipping_distance = 10000.f;

	while (app) // Application still alive?
	{
		// Using the align object, we block the application until a frameset is available
		rs2::frameset frameset = pipe.wait_for_frames();
		
		// rs2::pipeline::wait_for_frames() can replace the device it uses in case of device error or disconnection.
		// Since rs2::align is aligning depth to some other stream, we need to make sure that the stream was not changed
		//  after the call to wait_for_frames();
		if (profile_changed(pipe.get_active_profile().get_streams(), profile.get_streams()))
		{
			//If the profile was changed, update the align object, and also get the new device's depth scale
			profile = pipe.get_active_profile();
			align_to = find_stream_to_align(profile.get_streams());
			align = rs2::align(align_to);
			depth_scale = get_depth_scale(profile.get_device());
		}

		//Get processed aligned frame
		auto processed = align.process(frameset);

		// Trying to get both other and aligned depth frames
		rs2::video_frame other_frame = processed.first(align_to);
		rs2::depth_frame aligned_depth_frame = processed.get_depth_frame();
[... 131 lines cut ...]

	if (!depth_stream_found)
		throw std::runtime_error("No Depth stream available");

	if (align_to == RS2_STREAM_ANY)
		throw std::runtime_error("No stream found to align with Depth");

	return align_to;
}

bool profile_changed(const std::vector<rs2::stream_profile>& current, const std::vector<rs2::stream_profile>& prev)
{
	for (auto&& sp : prev)
	{
		//If previous profile is in current (maybe just added another)
		auto itr = std::find_if(std::begin(current), std::end(current), [&sp](const rs2::stream_profile& current_sp) { return sp.unique_id() == current_sp.unique_id(); });
		if (itr == std::end(current)) //If it previous stream wasn't found in current
		{
			return true;
		}
	}
	return false;
}

This is our code for aligning the RGB color image with the depth image and saving them. Hope to hear from you.

RealSense-Customer-Engineering, 2018-10-25:
[Realsense Customer Engineering Team Comment]
Hi @fu3lab,

I'm sorry for late.

Do you ask how to config the output of the RealSense camera?
If yes, you can replace

`rs2::pipeline_profile profile = pipe.start();`

by

`
    rs2::config cfg;
    cfg.enable_stream(rs2_stream::RS2_STREAM_DEPTH, 1280, 720, rs2_format::RS2_FORMAT_Z16);
    cfg.enable_stream(rs2_stream::RS2_STREAM_COLOR, 1280, 720, rs2_format::RS2_FORMAT_RGB8);
    rs2::pipeline_profile profile = pipe.start(cfg);
`

fu3lab, 2018-10-29:
Thank you very much. This is what we want.

closed 2018-10-30 by RealSense-Customer-Engineering

MRo47, 2019-09-16:
Thank You @RealSense-Customer-Engineering 

In case someone is looking for the code to use opencv instead of imgui.

`#include <opencv2/opencv.hpp>
#include <iostream>
#include <librealsense2/rs.hpp>
#include <cstring>
#include <algorithm>

float get_depth_scale(rs2::device dev)
{
  // Go over the device's sensors
  for (rs2::sensor& sensor : dev.query_sensors())
  {
    // Check if the sensor if a depth sensor
    if (rs2::depth_sensor dpt = rs2::depth_sensor(sensor))
    {
      return dpt.get_depth_scale();
    }
  }
  throw std::runtime_error("Device does not have a depth sensor");
}

void remove_background(rs2::video_frame& other_frame, const rs2::depth_frame& depth_frame, float depth_scale, float clipping_dist)
{
  const uint16_t* p_depth_frame = reinterpret_cast<const uint16_t*>(depth_frame.get_data());
  uint8_t* p_other_frame = reinterpret_cast<uint8_t*>(const_cast<void*>(other_frame.get_data()));

  int width = other_frame.get_width();
  int height = other_frame.get_height();
  int other_bpp = other_frame.get_bytes_per_pixel();

  //#pragma omp parallel for schedule(dynamic) //Using OpenMP to try to parallelise the loop
  for (int y = 0; y < height; y++)
  {
    auto depth_pixel_index = y * width;
    for (int x = 0; x < width; x++, ++depth_pixel_index)
    {
      // Get the depth value of the current pixel
      auto pixels_distance = depth_scale * p_depth_frame[depth_pixel_index];

  		// Check if the depth value is invalid (<=0) or greater than the threashold
  		if (pixels_distance <= 0.f || pixels_distance > clipping_dist)
  		{
  			// Calculate the offset in other frame's buffer to current pixel
  			auto offset = depth_pixel_index * other_bpp;

  			// Set pixel to "background" color (0x999999)
  			std::memset(&p_other_frame[offset], 0x01, other_bpp);
  		}
    }
  }
}

rs2_stream find_stream_to_align(const std::vector<rs2::stream_profile>& streams)
{
  //Given a vector of streams, we try to find a depth stream and another stream to align depth with.
  //We prioritize color streams to make the view look better.
  //If color is not available, we take another stream that (other than depth)
  rs2_stream align_to = RS2_STREAM_ANY;
  bool depth_stream_found = false;
  bool color_stream_found = false;

  for (rs2::stream_profile sp : streams)
  {
    rs2_stream profile_stream = sp.stream_type();
    if (profile_stream != RS2_STREAM_DEPTH)
    {
      if (!color_stream_found) //Prefer color
[... 121 lines cut ...]

    cv::cvtColor(image_frame, image_frame, CV_BGR2RGB);
    cv::namedWindow( "Image window", cv::WINDOW_AUTOSIZE ); // Create a window for display.
    cv::imshow( "Image window", image_frame );                // Show our image inside it.
    cv::namedWindow( "Depth window", cv::WINDOW_AUTOSIZE ); // Create a window for display.
    cv::imshow( "Depth window", image_depth );                // Show our image inside it.

    if((char)cv::waitKey(25)==27)
      break;
  }

  return EXIT_SUCCESS;

}
catch (const rs2::error & e)
{
  std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n " << e.what() << std::endl;
  return EXIT_FAILURE;
}
catch (const std::exception & e)
{
  std::cerr << e.what() << std::endl;
  return EXIT_FAILURE;
}
`

MRo47, 2019-09-16:
Thank You @RealSense-Customer-Engineering for the answer.
and @fu3lab for raising the issue.
If anyone is looking for the opencv compatible version , the code is below and is independent of any third party libraries from realsense.

[https://gist.github.com/MRo47/aecd9be1b58455f29b5c1544703589ae](url)

aalarcon96, 2019-10-22:
> Thank You @RealSense-Customer-Engineering for the answer.
> and @fu3lab for raising the issue.
> If anyone is looking for the opencv compatible version , the code is below and is independent of any [...]

The link for this is not working, is there another way to get this code?

Thank you!

MartyG-RealSense, 2019-10-22:
The link is here:

https://gist.github.com/MRo47/aecd9be1b58455f29b5c1544703589ae

You can also get it via this link if the above one stops working:

https://gist.github.com/MRo47

MRo47, 2019-10-22:
Thanks @MartyG-RealSense . @aalarcon96 hope that solved your problem. If not try copy pasting the link in your browser

aalarcon96, 2019-11-15:
Hello again, 

Is there a way to superimpose the opencv color and depth images like in rs-align? I have aligned color and depth frames using opencv but I was wondering if there was a way to view them in the same viewer like rs-align does. 

Thank you!

MRo47, 2019-11-16:
Hi @aalarcon96 
[https://docs.opencv.org/3.1.0/d5/dc4/tutorial_adding_images.html](url)
Maybe this helps?