realsenseai/librealsense #2553: Streaming RGB and Depth at the same time

upxela, 2018-10-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415  | 
| Firmware Version                   | 5.10.03 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.16.1      | 
| Language                            |  python2.7                           | 

### Issue Description
I'm currently using pyrealsense2 on python 2.7. 
I realized when I explicitly enable the depth and rgb stream on a configuration, I drop 25%-50% of my frames. However, if I run the pipeline on default parameters (e.g. no configuration), I obtain full framerate.

My end goal is to actually enable depth, rgb, and a single IR stream, so I have to enable the streams explicitly on a config. Why does enabling these streams drop frames?

upxela, 2018-10-18:
@RealSense-Customer-Engineering Thanks for getting back to me.

1. My desired config would be set as the following:
```
config = rs.config()
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
config.enable_stream(rs.stream.infrared, 1280, 720, rs.format.y8, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.rgb8, 30)
config.enable_record_to_file('path2bag')
```
2. A USB3 port is indeed being used, and the D435 (I mistakenly wrote D415 on the issue) is recognized as a USB3 device.
3. The way I verify framerate is checking the number of image messages in the bag file.  The opencv_viewer_example.py doesn't look like it's dropping too many frames, but the only sure way to find that out is by calculating expected number of frames and comparing that to actual number of frames.
3. Using realsense-viewer also fails to give me 100% of the expected number of frames.

RealSense-Customer-Engineering, 2018-10-19:
[Realsense Customer Engineering Team Comment]
Hi @upxela,

The system overhead and software stack may affect the stream fps and record. See #2563 as a similar issue. Will further check if any optimization there.

closed 2019-02-19 by RealSenseCustomerSupport