realsenseai/librealsense #2554: Question about Frame Sync Index [D435][HW sync]

shane7226107, 2018-10-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435} | 
| Firmware Version                   | 5.10.3 | 
| Operating System & Version | Win 10 | 
| Kernel Version (Linux Only)    |  -         | 
| Platform                                 | PC |
| SDK Version                            |  2.16.1                          | 
| Language                            |  C | 
| Segment			|  VR/AR                        | 

### Issue Description
Hello guys,
I am using two D435 and connected their HW sync pin as the [whitepaper](https://realsense.intel.com/wp-content/uploads/sites/63/Multiple_Camera_WhitePaper04.pdf) instructed:
[image]

The SW seems working, both camera supports RS2_OPTION_INTER_CAM_SYNC_MODE and I can setup one camera as Master and the other as Slave:

`bool support_hw_sync = depth_sensor.supports(RS2_OPTION_INTER_CAM_SYNC_MODE);`
`if (support_hw_sync) {`
`float sync_mode = found_master ? (float)1.0 : (float)2.0 ;// [0:default/1:sync master/2:sync slave]`
`depth_sensor.set_option( RS2_OPTION_INTER_CAM_SYNC_MODE, sync_mode );`

Here comes the problem when I want to verified the hw sync result.

The way is using some information as "frame sync index", with same "frame sync index" , the content captured by different camera should be the same.

I have tried the following metadatas as "frame sync index":

**[1] RS2_FRAME_METADATA_FRAME_TIMESTAMP or RS2_FRAME_METADATA_SENSOR_TIMESTAMP**
    According to it is stated here (https://github.com/IntelRealSense/librealsense/issues/2188)
    These timestamp are counted by individual device clock, which means they are not aligned at all.
    By my experiments, it is true that when HW sync mode is on, the same frameset polled at the same time from different camera,
    the FRAME_TIMESTAMP and SENSOR_TIMESTAMP are not aligned
    => it can not be used as "frame sync index".

**[2] RS2_FRAME_METADATA_FRAME_COUNTER** 
    The whitepaper says that with hw sync, the counters are aligned between cameras;
    However, according to my experiment, they are not:
// camera 1, serial number 825312070233, polled at 30:813, frame counter is 172
[10/17/18 16:35:30:813][8152 018664][D] [825312070233] polled frame[172] timestamp (1539765330757.000000)
// camera 2, serial number 819312070204, polled at 30:814, frame counter is  215
[10/17/18 16:35:30:814][8152 018664][D] [819312070204] polled frame[215] timestamp (1539765330756.000000)

=>the frame counters are NOT aligned and not useful

**[3] RS2_FRAME_METADATA_BACKEND_TIMESTAMP**
    So I turn to BACKEND_TIMESTAMP. 
    At first glance, it seems doable if I use this timestamp as reference, the framesets acquired from rs2::pipelines are perfectly aligned (by checking the IR image shooting a same stopwatch)
[image]
[image]

Unfortunately, with more test, I have noticed that sometimes the BACKEND_TIMESTAMP are same even the frames are polled at different time:

// camera 1, serial number 825312070233, polled at 30:877, backend timestamp is 1539765330821
[10/17/18 16:35:30:877][8152 018664][D] [825312070233] polled frame[174] timestamp (1539765330821.000000) 
// camera 1, serial number 825312070233, polled at 30:937, backend timestamp is 1539765330821
[10/17/18 16:35:30:937][8152 018664][D] [825312070233] polled frame[175] timestamp (1539765330821.000000)
// camera 1, serial number 825312070233, polled at 30:940, backend timestamp is 1539765330821
[10/17/18 16:35:30:940][8152 018664][D] [825312070233] polled frame[176] timestamp (1539765330821.000000)

There are 3 consecutive frames with the same BACKEND_TIMESTAMP!

My question is, is there any information that I can used as "frame sync index"? 

Or is there any official way to make sure the framesets acquired at the same time are captured exactly at the same time?

Thank you!

RealSense-Customer-Engineering, 2018-10-23:
[Realsense Customer Engineering Team Comment]
Hi @shane7226107,

Backend timestamp could be used as the reference.
What is the resolution and frame rate, and streams that are configured when the test was performed?

Regards,
Yusheng

shane7226107, 2018-10-24:
Hi Yusheng,

I enabled Color, Depth and IR left stream,

The resolution resolution of them are all  640x480 at frame rate 30fps.

Thanks,
Shane

RealSense-Customer-Engineering, 2018-10-31:
[Realsense Customer Engineering Team Comment]
Hi Shane,
This might be related to the Windows OS as I have not yet seen the repeated backend timestamps for different frame ids on Linux. I will test it on Win 10 later.
Is it possible for you try the same code on a Linux host?

shane7226107, 2018-10-31:
Hi RealSense,

Sorry I can not test the same code on a Linux host, my workstation are all windows environments.

Thanks

trenneman, 2018-10-31:
We have the same issue of multiple consecutive frames with the same BACKEND_TIMESTAMP. Also on Windows 10. RGB (full HD), IR (640x480) and depth (640x480) streams running at 30 fps.

trenneman, 2018-11-09:
I've now tested the backend timestamp on 4 different installations:
✔️ Ubuntu 18.04 *
✔️ Windows 10, version 1803
❌ Windows 10, version 1809 *
❌ Windows 10, version 1809

@shane7226107 What Windows version are you running?

*) Same machine

shane7226107, 2018-11-16:
Hi Trenneman,

my windows version is Win 10 64 bits ,version 1803, build 17134.407

RealSense-Customer-Engineering, 2019-01-11:
[Realsense Customer Engineering Team Comment]
Hi @shane7226107,
The issue seems to be related to the Windows metadata support.
Please have a look at this whitepaper, and to the section 2.F
https://realsense.intel.com/wp-content/uploads/sites/63/Multiple_Camera_WhitePaper_rev1.1.pdf

It will lead you to the following page and instruct how to properly enable metadata on Windows.
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

closed 2019-02-02 by RealSenseCustomerSupport