realsenseai/librealsense #2555: example.hpp splitting OpenGL window into classes!

HippoEug, 2018-10-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version | Win 10 | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC |
| SDK Version                            | 2.16.1 | 
| Language                            | C++  | 
| Segment			| others                   | 

Hello once more again!

I previously attempted to create a new class file to create a OpenGL window to display the camera frames without example.hpp [here](https://github.com/IntelRealSense/librealsense/issues/2504).

But I still had trouble doing it despite, so I went back to looking at example.hpp and modified as little as possible to make it work. However, I have another slight problem that I cannot seem to overcome.

### Desired Output
[image]

Here is the source code to display the above:
```
// main.cpp

int main(int argc, char* argv[]) {
	rsgui rsgui;
	rsgui.init();
}
```
```
// rsgui.hpp

class rsgui {
public:
	window_rs window; // window_rs is actually "window" class from example.hpp, just slightly modified.

	void init();
};
```
```
// rsgui.cpp

void rsgui::init() {
	window.onCreate(1280, 720, "IntelRealsenseXOneS"); //onCreate is actually constructor of "window" class from example.hpp
	ImGui_ImplGlfw_Init(window, false);

	texture_rs rgbPreview; // "texture" class from example.hpp
	rs2::pipeline pipe;

	pipe.start();
	
	bool show_color_camera = false;
	bool button_pressed = false;
	
	while (window) {
		rs2::frameset data = pipe.wait_for_frames();
		rs2::frame color = data.get_color_frame();
		rgbPreview.render(color, { 0, 0, window.width() / 2, window.height() / 2 });

		ImGui_ImplGlfw_NewFrame(1);
		menuGUIx(show_color_camera, button_pressed);
	}
	
}
```

## Issue
As you noticed in `rsgui.hpp`, I created a public class object `window_rs window;` so as to be able to access window in the other class functions, instead of just using it in `init()` for example. I hope to be able to do this in order to make the code more modular and simpler. Also, there is a `while(1)` loop in init, which is a blocking function :( ..

Here is the source code I have tried:
```
// main.cpp

int main(int argc, char* argv[]) {
	rsgui rsgui;

	rsgui.init(); // Initializing the window object in rsgui.hpp class
	std::thread t2(&rsgui::testFunc, rsgui); // Threading to display camera frames

	t2.join();
}
```
```
// rsgui.hpp

class rsgui {
public:
	window_rs window;

	void init();
	void testFunc();
};
```
```
// rsgui.cpp

void rsgui::init() {
	window.onCreate(1280, 720, "IntelRealsenseXOneS");
	ImGui_ImplGlfw_Init(window, false);
}

void rsgui::testFunc() {
	texture_rs rgbPreview;
	rs2::pipeline pipe;

	pipe.start();

	bool show_color_camera = false;
	bool button_pressed = false;

	while (window) { // PROBLEM(?)
		rs2::frameset data = pipe.wait_for_frames();
		rs2::frame color = data.get_color_frame();
		rgbPreview.render(color, { 0, 0, window.width() / 2, window.height() / 2 });

		//ImGui_ImplGlfw_NewFrame(1); // Commented out since testing 
		//menuGUIx(show_color_camera, button_pressed); // Commented out since testing
	}
}
```
The code does compile, but I get this result:
[image]

Through some testing, I believe it is `while(window)` in `rsgui.cpp` that is causing the issue, since `while(1)` also causes the issue. 

In case you want to see my modified `example.hpp` file, the source code for that class is [here](https://hastebin.com/yetuyuvoya.cpp).

## Any help is appreciated, thank you once again for reading! Let me know if I need to further clarify on anything 😄

HippoEug, 2018-10-25:
No I have not.. because I couldn't find a solution I ended up writing all my test functions in a single function `rsgui.init()`, all together in a `while(window)` loop. 

Here is what it currently looks like:
#### `Source.cpp`
```
int main(int argc, char* argv[]) {
	rsgui rsgui;

	std::thread t1(&rsgui::init, rsgui);
	t1.join();
}
```

#### `rsgui.cpp`
```
void rsgui::init() {
	window.onCreate(1280, 720, "IntelRealsenseXOneS");
	ImGui_ImplGlfw_Init(window, false);

	std::ofstream testFile;
	testFile.open("test.csv");

	texture_rs rgbPreview, depthPreview;
	rs2::colorizer color_map;
	//rs2::pipeline pipe;
	auto pipe = std::make_shared<rs2::pipeline>();

	pipe->start(); // Cannot loop this over and over again

	bool show_color_camera = false;
	bool show_depth_camera = false;
	bool camera_button = false;
	bool coordinates_button = false;
	bool CSV_button = false;

	bool recorded = false;
	bool recording = false;

	while (window) {
		ImGui_ImplGlfw_NewFrame(1);
		menuGUI(show_color_camera, show_depth_camera, camera_button, coordinates_button, CSV_button);

		if (show_color_camera == true || show_depth_camera == true) {
			rs2::frameset data = pipe->wait_for_frames();
			rs2::frame color = data.get_color_frame();
			rs2::frame depth = color_map(data.get_depth_frame());
			//rs2::depth_frame depthS = data.get_depth_frame();

			if (show_color_camera) {
				//std::cout << depthS.get_distance(640, 360) << std::endl;
				rgbPreview.render(color, { 640, 0, window.width() / 2, window.height() / 2 });
			}
...
...
```

I will give it another shot to try and make it work again tomorrow. If you need any further information please let me know! Thank you Sergey!

RealSense-Customer-Engineering, 2018-11-02:
[Realsense Customer Engineering Team Comment]
Hi @HippoEug,

In the GLFW document, we can find these words.
`Most GLFW functions must only be called from the main thread, but some may be called from any thread.`
http://www.glfw.org/docs/latest/intro_guide.html#thread_safety

I think, write all GLFW function in the same thread is an easy solution.

HippoEug, 2018-11-05:
Yeah I think that is my only option at the only time. Thank you for your reply ❤️!