realsenseai/librealsense #2556: multicam live reconstruction

EmmanuelCos, 2018-10-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC  |
| Language                            |  C++   | 

Hello,

I am using two cameras at the same time with the multicam feature. I would like to make a live reconstruction, and not display two views of each camera but display only one view with a reconstruction of what both camera detect. I could transmit the angle difference between the two cameras to falicitate the reconstruction.
Has someone already implemented that kind of feature?

Best regards,

Emmanuel

MartyG-RealSense, 2018-10-18:
In January 2018 at the Sundance festival, Intel demonstrated 'volumetric capture' 180 degree scanning of the human body with four hardware-synced D435s.  

According to an Intel blog article about the demo, they took point cloud scans from four separate PCs (one for each camera) and used a 5th PC to sync it all together.

https://realsense.intel.com/intel-realsense-volumetric-capture/

My transcript of Intel's recent webinar about multiple cameras has suggestions about how to approach combining point clouds into a single one - by using the software Vicalib, or by doing an Affine Transform operation to rotate the clouds so they can be appended together into a single cloud.

https://communities.intel.com/message/566337#566337

dorodnic, 2018-10-18:
Also please take a look at #2531. I know this issue is Windows and ROS is (mostly) Ubuntu, but this can give you some direction (and perhaps you could even make it work with newly announced ROS for Windows (#2465)

RealSense-Customer-Engineering, 2019-01-04:
[Realsense Customer Engineering Team Comment]
Closing this now.
 Feel free to reopen if your issue or problem was not addressed.

closed 2019-01-04 by RealSense-Customer-Engineering