realsenseai/librealsense #2560: align color to depth

toinsson, 2018-10-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version | Ubuntu 18.04  | 
| Kernel Version (Linux Only)    | 4.15.0-36-generic | 
| Platform                                 | |
| SDK Version                            | 2.x - from apt | 
| Language                            | python | 
| Segment			| | 

### Issue Description
I have some trouble getting the depth aligned color data.
This code:
```
import matplotlib.pyplot as plt
import numpy as np
import pyrealsense2 as rs
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
profile = pipeline.start(config)
frames = pipeline.wait_for_frames()
align_to = rs.stream.depth
align = rs.align(align_to)
aligned_frames = align.process(frames)
fig, ax = plt.subplots(1,2, figsize=(12,12))
ax[0].imshow(np.asanyarray(aligned_frames.get_color_frame().get_data()))
colorizer = rs.colorizer()
ax[1].imshow(np.asanyarray(colorizer.colorize(aligned_frames.get_depth_frame()).get_data()))
```

produces:

[image]

And in case the config does not enforce the same resolution for both stream, then `align.process` does crash.

The rs-pointcloud example does work perfectly though.

dorodnic, 2018-10-18:
Hi @toinsson 
I can't see anything immediately wrong with your code.
I did try to run the [distance to object [image]](https://mybinder.org/v2/gh/IntelRealSense/librealsense/jupyter?filepath=notebooks/distance_to_object.ipynb) notebook, and replaced **Stream Alignment** section with:
```python
# Create alignment primitive with **depth** as its target stream:
align = rs.align(rs.stream.depth)
frameset = align.process(frameset)

# Update color and depth frames:
aligned_color_frame = np.asanyarray(frameset.get_color_frame().get_data())

# Show the two frames together:
images = np.hstack((aligned_color_frame, colorized_depth))
plt.imshow(images)
```
[image]

Maybe something in this example can help you find the problem...

toinsson, 2018-10-18:
thanks for the quick reply. It is the same code indeed. I'll do a fresh reinstall and see if this problem disappears.

toinsson, 2018-10-18:
Seems like the issue was in pyrealsense2. Was using v2.16.1.284. Move up to v2.16.1.291 and it works now. Sorry for the noise.

closed 2018-10-18 by toinsson

jaimeet64, 2021-06-07:
@dorodnic  I have tried the example that you have mentioned above. I am pointing my camera to a flat surface and I have corresponding RGB plus depth. Check out the image uploaded below. 

[image]

Later I replace the whole depth value with a mean value as the depth is supposed to be constant for a flat surface.  When I calculate "aligned_color_frame", I get the following result.

[image]

I don't know why I am still getting holes in the aligned_color_frame even though the depth has no holes? 

I will add my code snippet below:

# Stream Alignment

# Create alignment primitive with color as its target stream:
align = rs.align(rs.stream.color)
frameset = align.process(frameset)

# Update color and depth frames:
aligned_depth_frame = frameset.get_depth_frame()
colorized_depth = np.asanyarray(colorizer.colorize(aligned_depth_frame).get_data())
mean_depth = np.mean(colorized_depth)
print(mean_depth)
#colorized_depth[colorized_depth >= 0] = mean_depth
# Show the two frames together:
images = np.hstack((color, colorized_depth))
plt.imshow(images)