realsenseai/librealsense #2562: Reconfigure callback failed with exception start_streaming(...) failed for realsense camera d435

Anitawuxy, 2018-10-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 5.10.6 | 
| Operating System & Version |   linux (Ubuntu 14)  | 
| Kernel Version (Linux Only)    |  4.4                                         | 
| Platform                                 | PC  |
| SDK Version                            |  Intel® RealSense™ SDK 2.0                      | 
| Language                            |  C/C#/labview/nodejs/opencv/pcl/python/unity                           | 
| Segment			|  Robot                         | 

### Issue Description
The rs_camera.launch and rs_rgb.launch don't work. The error message is [ WARN] [1539877162.228800976]: Reconfigure callback failed with exception start_streaming(...) failed. UVC device is already streaming! I just updated the realsense firmware to the latest version 5.10.6. 
The launch command is:
[image]
[image]
[image]
I also checked the lsusb:
[image]
In addition, the camera is connected to USB 3.0.

Thanks in advance

SCH923, 2018-10-19:
I had same problem.  realsense firmware to the latest version 5.10.6

Anitawuxy, 2018-10-22:
@SCH923 Do you find any solution then?

Anitawuxy, 2018-10-22:
Also when I opened RVIZ, the global status has error. 
[image]

SCH923, 2018-10-22:
That warning still appears.
But I realized that was not problem.

@Anitawuxy
1.
change map to other 

Grobal Option
->Fixed Frame map

2.
Click Add 

3.
'By topic ' tab
Choice whatever you want. :>

j-polden, 2018-10-30:
Any updates on this error? We are experiencing it as well.

Anitawuxy, 2018-10-30:
The warning is still there. But as SCH925 said, it doesn’t matter. For my case, I just launched rs_rgb.launch. So it is in order that my rvit doesn‘t have /map topic.

closed 2018-10-30 by Anitawuxy